OrcaSlicer/OrcaSlicer #8561: 2.3.0 BETA buld plate names and multiple build plates not saving

BigToyBox, 2025-02-25:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 BETA

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

### Printer

Sovol SV07

### How to reproduce

Try to save a single build plate with a name.

Try to save a multiple build plate project

When returning to the project, the bug appears!

### Actual results

This was a four plate project with one part per plate and each plate named.

[image]

### Expected results

It should work as expected

### Project file & Debug log uploads

[SICA plate bug.zip](https://github.com/user-attachments/files/18967079/SICA.plate.bug.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Noisyfox, 2025-03-01:
> Try to save a single build plate with a name.

Sorry I'm not quite sure what this means. Could you provide more information like what steps did you perform?

Truncated, 2025-03-11:
This is what I was referring to with https://github.com/SoftFever/OrcaSlicer/issues/8302#issuecomment-2712466494

I'm doing this all in Windows 10.

**Steps to recreate:**
1. Create a new project in Orca (I've done this in 2.3.0 Beta2 and the nightly build RC from today)
2. Add a second plate to the project.
3. Import a primitive / any model to each plate (I added an Orca Cube from the Handy Models)
4. Change the plate names to unique titles.
5. Save the project & close Orca.
6. Reopen Orca and open the saved project.

**The expected bug experience is:**
- You will find only one plate existing with "Untitled" as the name. The added models are floating, generally spaced where plates should be, without colors or associated customizations.

**Inconsistently, I get this as well:**
- The opened file name is listed as "Untitled" instead of the file name you just gave it.
- When this happens, you can't use save, as it sees it as a new file. If you save-as overwriting the test file in the example above, it will update plate names.
---
I compared Bambu Studio 1.10.1.50 and Orca Slicer 2.3.0 Beta 2 / RC with the following steps in each:

1. Created a new project.
2. Added second plate.
3. Named the plates "Test Plate 1" and "Test Plate 2" in both.
4. Added a primitive cube to the first plate.
5. Added a primitive cylinder to the second plate.
6. Saved the files as Bambu 1-10-1-50 2plate Test.3mf and Orca Beta 2-RC 2plate Test.3mf respectively.

- Bambu Studio can open the Bambu file no problems.
- Bambu Studio can open the Orca file - model is there but second plate and names are missing.
- Orca can open both files, but both are missing the second plate and names.

I then opened both in Orca (both broken) and saved them as: 

1. Orca Opened and Saved Bambu 1-10-1-50 2plate Test.3mf
2. Orca Opened and Saved Orca Beta 2-RC 2plate Test.3mf

It looks like whatever it's dropping on load is compounding in subsequent saving, deteriorating the resulting files.

I included my copies of these second grouping of test files, along with screenshots in both explorer with preview mode and the apps.

[OrcaTesting.7z.zip](https://github.com/user-attachments/files/19175684/OrcaTesting.7z.zip)

Truncated, 2025-03-11:
Given these file issues, i had to personally revert to 2.2.0, so I went ahead and ran some additional comparison tests while I was at it.
**Steps Performed:**

1. Uninstalled Beta 2/RC
2. Installed 2.2.0 Release.
3. Downloaded Beta 1 portable.
4. Opened files with 2.2.0 and Beta 1 Portable; saved them each with clear names in relation to each other.
5. Created a baseline 2Plate test file from both versions to compare.

When I opened the Orca Beta 2-RC 2plate Test.3mf in version 2.2.0, I got this message:
[image]
But opened with multiple plates and names!

**My current logic findings:**

- None of the beta versions opened any of these multi-plate files correctly, so something is impacting the parsing of the file on open.
- The first save of a brand new project from all 2.3.0 Beta versions is saved in a way that doesn't invalidate it with Orca 2.2.0, but with Bambu Studio, so saving a file is malformed in some way from the start.
- Subsequent openings and saves compounds these errors. I expect this will cause files to further deterioriate as they are opened and saved again.

All of my versions from the tests above to compare:

[OrcaTesting with more version tests.zip](https://github.com/user-attachments/files/19176230/OrcaTesting.with.more.version.tests.zip)

Noisyfox, 2025-03-11:
[image]

Please make sure this option is not set to "Load Geometry Only"

yangxyz, 2025-04-19:
> When you click Add, Orca asks if I always want the Geometry only in the future. When I said Yes, thinking it was for Add, Orca sets the Project Load Behavior option which applies to **_both_** the Open and Add menu options.

Bang on yep this is what's causing the bug. Switching my preference to "ask when relevant" has fixed the issue for me.

VidiotGeek, 2025-05-31:
I just went through this same issue. Glad the "ask when relevant" setting fixed the problem but absolutely agree that "open" and "add" are 2 fundamentally different things. The fact that 3mf is both a project file and object distribution file feels inherently confusing to me.

Noisyfox, 2025-06-01:
We've now removed the "don't ask again" checkbox so people won't accidentally change this behavior without knowing.

Noisyfox, 2025-06-01:
> The fact that 3mf is both a project file and object distribution file feels inherently confusing to me.

Yes that's the fundamental problem, unfortunately cannot be solved.

closed 2025-09-08 by github-actions[bot]