OrcaSlicer/OrcaSlicer #8563: Bridge on TOP layer aroud holes

locki-cz, 2025-02-25:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

from 2.2.0 above

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Voron 2.4

### How to reproduce

Hi, this issue in Orca Slicer has been present since version 2.2.0, which was released four months ago. Because of this, I’m still using version 2.1.1, where the problem doesn’t occur. :(

How to reproduce it:
An object with threaded holes (size doesn’t matter)
Layer height of 0.2 mm or more (it doesn’t appear at 0.15 mm or lower)
A single perimeter on the top layer

### Actual results

[image]

[image]

### Expected results

No bridges on top layer

### Project file & Debug log uploads

[orca-test-bridge-top-layer.zip](https://github.com/user-attachments/files/18967603/orca-test-bridge-top-layer.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

discip, 2025-02-26:
Seems to be related to **`Layer height`**, as the problem disappears as soon as this value is reduced to at least 1.5 mm.

discip, 2025-02-26:
Seems to be related to **`Layer height`**, as the issue disappears as soon as this value is reduced to at least 1.5 mm.

locki-cz, 2025-02-26:
> Seems to be related to **`Layer height`**, as the issue disappears as soon as this value is reduced to at least 1.5 mm.

0.15 mm ;) And it is weird that nobody noticed it before. Hopefully, it can be easy to fix if it is only on a higher layer height.

discip, 2025-02-26:
> 0.15 mm ;)

Yeah, that's what I actually meant. 😊

locki-cz, 2025-02-27:
@discip will it be easy to fix?

tetsu97, 2025-02-28:
Same problem

Ataraxia-Mechanica, 2025-03-03:
> Seems to be related to **`Layer height`**, as the issue disappears as soon as this value is reduced to at least 0.15 mm.

With sharp enough edges the problem still exists. This is with 0.1mm layer height:

[image]

locki-cz, 2025-03-11:
Can please somebody look on this?

discip, 2025-03-11:
> [@discip](https://github.com/discip) will it be easy to fix?

Sorry, I forgot to answer.
Unfortunately I'm no programmer, so no idea. 😅

locki-cz, 2025-03-11:
@Noisyfox can you please look on this bug?

daniiooo, 2025-03-17:
I believe I've encountered the same bug. In my case additional bridge is getting generated at certain layer heights, not on top layer though. For my model the issue appears at heights 0.12 and 0.10. The issue doesn't occur if I slice the model with 0.12 height right after opening orcaslicer. If I slice it again the unwanted bridge appears.

Version: 2.3.0-rc

Model: 
[bridge_problem.zip](https://github.com/user-attachments/files/19299416/bridge_problem.zip)

[image]

Ataraxia-Mechanica, 2025-03-18:
> I believe I've encountered the same bug. In my case additional bridge is getting generated at certain layer heights, not on top layer though. For my model the issue appears at heights 0.12 and 0.10. [...]

Your model is non-manifold which caused this problem. If you are on windows you can click the repair button.

[image]

daniiooo, 2025-03-18:
Ah I didn't notice the model is borked yesterday. I'll be able to see if repairing it will fix the issue later today. When I saw the problem happening only on certain layer heights I didn't think it could be an issue with the model, thanks for pointing that out.

locki-cz, 2025-03-21:
Still in 2.3.0 final version

locki-cz, 2025-03-21:
probably related to this change: [https://github.com/SoftFever/OrcaSlicer/pull/6654](https://github.com/SoftFever/OrcaSlicer/pull/6654)

same problem in latest Prusaslicer 2.9.1

[image]

locki-cz, 2025-03-23:
@Noisyfox Do you think it would be possible to revert it to its previous state before impementing: [https://github.com/SoftFever/OrcaSlicer/pull/6654](https://github.com/SoftFever/OrcaSlicer/pull/6654)

Noisyfox, 2025-03-23:
Reverting that causes poor bridge anchoring if top layer is also the bridge layer.
The real problem is, what do we want to do when the top layer contains bridges. I don't think there is a simple answer to this tbh.

Noisyfox, 2025-03-24:
Please give the new pr a try.

locki-cz, 2025-03-24:
> Please give the new pr a try.

The fix works, tested, awesome—I didn't even hope for such a quick fix! Hopefully, it didn't break anything else. Thanks a lot!

Noisyfox, 2025-03-25:
> Hopefully, it didn't break anything else

That's also what I'm worried about. Gonna do more tests.

locki-cz, 2025-03-26:
> > Hopefully, it didn't break anything else
> 
> That's also what I'm worried about. Gonna do more tests.

Using that and not found any problems with bridges ... so it looks ok! :)

Ataraxia-Mechanica, 2025-03-29:
> The fix works, tested, awesome—I didn't even hope for such a quick fix! Hopefully, it didn't break anything else. Thanks a lot!

I have unfortunate news. This broke thin top surfaces.

[image]

[image]

locki-cz, 2025-03-29:
> I have unfortunate news. This broke thin top surfaces.
> 
> [image]
> 
> [image]

Can you share that stl file?

Ataraxia-Mechanica, 2025-03-29:
> > I have unfortunate news. This broke thin top surfaces.
> Can you share that stl file?

I've posted that in the pull request page, here's the file: [file](https://github.com/user-attachments/files/19521139/Printed_20-4040_Blind_Joint_Corner_Jig.zip)

locki-cz, 2025-03-29:
[image]

Ataraxia-Mechanica, 2025-03-30:
> [image]

My file uses 0.6mm nozzle which would make the thin top surface narrow enough and break it.

locki-cz, 2025-03-30:
> > [image]
> 
> My file uses 0.6mm nozzle which would make the thin top surface narrow enough and break it.

[image] also with 0.6 nozzle with new fix: [](https://github.com/SoftFever/OrcaSlicer/actions/runs/14147293451)

Ataraxia-Mechanica, 2025-03-30:
> also with 0.6 nozzle with new fix: https://github.com/SoftFever/OrcaSlicer/actions/runs/14147293451

Just installed it, it's also fixed on my side now. Nice.

[image]

commit 3a114a5d (2025-04-01): Avoid creating bridges on very tiny surface cracks (#9053)

Fix #8563

Before:
[image]

After:
[image]
PR #9053 (2025-04-01): Avoid creating bridges on very tiny surface cracks
Fix #8563

Before:
[image]

After:
[image]
closed 2025-04-01 by SoftFever