OrcaSlicer/OrcaSlicer #8567: auto arrange crashes orca

zubamse, 2025-02-25:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

11

### Additional system information

Intel® Core™ i9-13900HX, 36 MB cache, 24 cores
32 GB, 2 x 16 GB, DDR5, 4800 MHz, non-ECC, Dual Channel
NVIDIA® GeForce RTX™ 4080, 12 GB GDDR6

### Printer

Bambu P1S

### How to reproduce

open any file
click on arrange
now it crashes

https://youtu.be/HyTWYSIMhI4

### Actual results

I'm pretty sure its a me problem, but i have reinstalled the program and older versions, and it still happens

### Expected results

not to crash, works on my other "older" laptop 

### Project file & Debug log uploads

[crash_Tue_Feb_25_21_14_47_0.log](https://github.com/user-attachments/files/18972009/crash_Tue_Feb_25_21_14_47_0.log)

### Checklist of files to include

- [x] Log file
- [ ] Project file

klylabs, 2025-02-26:
can you provide the project for debugging?

TranceAerospace, 2025-02-27:
Had this happen intermittently as well, once when adding new filaments and another when clicking the slice button. Windows 11 machine

zubamse, 2025-02-28:
> can you provide the project file for debugging?

it doesnt matter the file, it happens to all files.
Also with other printers

X-sam, 2025-03-31:
I'm having a similar issue- seems to be specifically if you have two plates set to "by object" and it can't arrange them all on one. Attached minimal 3mf that reproduces the issue.

[test.zip](https://github.com/user-attachments/files/19538363/test.zip)

zubamse, 2025-04-12:
Looks like it happens when using the fast G-code from this site, and does not happen when using stock g-code.

https://www.printables.com/model/344496-fast-start-gcode-improved-quick-bed-leveling-end-r

closed 2025-07-19 by github-actions[bot]