OrcaSlicer/OrcaSlicer #8572: M106 P3 S178 after PRINT_END

zerodameaon, 2025-02-26:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-Beta

### Operating System (OS)

Windows

### OS Version

WIndows 10

### Printer

Voron 2.4

### How to reproduce

1: Slice file
2: Send to printer
3: Wait for print job to finish

### Actual results

GCODE that is produced in 2.3.0-Beta has the following GCODE. The part cooling fan goes to 70% after the print finishes.

EXCLUDE_OBJECT_END NAME=Multiboard_Cylinder_Holder_45mm_id_0_copy_0
M106 S0
;TYPE:Custom
; filament end gcode 
M106 P3 S0
PRINT_END
**M106 P3 S178**
M73 P100 R0
; EXECUTABLE_BLOCK_END

### Expected results

GCODE produced with 2.2.0 does not have the M106 P3 S178 and the print cooling fan turns off.

EXCLUDE_OBJECT_END NAME=tower_with_seam.stl_id_0_copy_0
M106 S0
;TYPE:Custom
; filament end gcode 
PRINT_END
M73 P100 R0
; EXECUTABLE_BLOCK_END

### Project file & Debug log uploads

[ORCA GCODE.zip](https://github.com/user-attachments/files/18977401/ORCA.GCODE.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

wight554, 2025-02-26:
I've got this issue too after upgrade
Same gcode is also added after START_PRINT:

[image]

wight554, 2025-02-26:
[image]
Oh I see, these are enabled by default

zerodameaon, 2025-02-26:
I missed that under filament. I did not think to look there for something like this. It should be off by default for for all the filaments since I now have to go turn that off for a lot of them. Or maybe it should be moved under the machine profile.

zerodameaon, 2025-02-26:
I need to check, but that may override the cooling settings of the part cooling fan if it is set during a print. 

I missed that option so yes, per filament off by default.

wight554, 2025-02-26:
> I need to check, but that may override the cooling settings of the part cooling fan if it is set during a print.
> 
> I missed that option so yes, per filament off by default.

Wiki has nice override macro, so not a problem
https://github.com/SoftFever/OrcaSlicer/wiki/air-filtration
Just not enough warning about using the feature considering it's on by default

EricZimmerman, 2025-03-06:
this still happens with with the exhaust fans unchecked

Snelso91, 2025-03-21:
I've noticed this issue too.
I don't understand why the default setting in the filament profile is to leave the exhaust fan on at 70% after the print has finished, instead of just turning it off. Not only is this not needed (as clearly demonstrated by the default behaviour of Bambu PLA or Bambu PETG on my P1S, which only have the exhaust fan on during the print), but it also means that the fan stays on infinitely, requiring user intervention to turn it off after every print, which is even more useless...

Also, it seems the reason why this is only occurring/being noticed more now is because a lot of the base/default filament profiles have been changed from 2.2.0 to 2.3.0 to have the "activate air filtration" checkbox enabled instead of disabled like it used to be, and having the checkbox enabled by default is incorrect for a lot of filaments, as I outlined in my related issue #9002.
For me at least, the short term fix is to simply disable the exahust fan checkbox and then save the filament profile, and then everything goes back to how it was in 2.2.0, but I'm assuming this isn't a fix for everyone, as @EricZimmerman said.

EricZimmerman, 2025-03-21:
I unchecked it and it still emits the fan code.

zerodameaon, 2025-03-21:
You just reminded me to check this, it appears to be fixed on the 2.3.0-rc. None of my profiles have it on now where every one had it on in the 2.3.0 betas. Is anyone else seeing that behavior?

zerodameaon, 2025-03-21:
I missed that the official release came out yesterday, Orca didn't prompt me this morning. It appears that it is also off on the official release for me.

Snelso91, 2025-03-21:
> Is anyone else seeing that behavior?

Unfortunately in my case, it seems that the air filtration checkbox is still on for a lot of Bambu filaments in 2.3.0 official release (such as ABS), so I still get the exhaust fan turned on during and after the print by default.
I'm guessing someone messed up the base/parent filament profile from which all the individual Bambu filament profiles are derived, which is why it's happening. At least for now though, I can simply override the default by unchecking the box and saving it as a new filament profile based on the default ones.

GoMachV, 2025-04-17:
ok, sorry but I am new to the way things work here. I have this same issue, but it appears nobody is looking into it or fixing it? Is it just a "tough luck" sort of thing? Or am I missing something? The workaround doesnt work for me, as it resets the filaments every time I sync the ams, and with several machines and AMS units i"m syncing all the time. I assume adding end gcode wont help as it is overwritten?

Snelso91, 2025-04-21:
> ok, sorry but I am new to the way things work here. I have this same issue, but it appears nobody is looking into it or fixing it? Is it just a "tough luck" sort of thing? Or am I missing something?

@GoMachV please see my comment here about this: https://github.com/SoftFever/OrcaSlicer/issues/9002#issuecomment-2819193016

retro486, 2025-09-28:
Unchecking the air filtration box in the filament settings removes the final M106 in 2.3.1-beta. It's also not default-enabled in the majority of generic filament profiles as far as I can see...

GoMachV, 2025-09-29:
the problem is you can uncheck it and save as a new filament but every time you sync it reverts back to the filament in the ams. I cant believe they aren't willing to fix something so blatantly wrong when the fix seems so incredibly easy.