OrcaSlicer/OrcaSlicer #8578: Some part of support is sliced in the middle of the air

Felixoid, 2025-02-26:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

Arch Linux

### Printer

Anycubic Kobra 3

### How to reproduce

[laptop-stand-hooks.zip](https://github.com/user-attachments/files/18981435/laptop-stand-hooks.zip)

1. Open the laptop-stand-hooks.3mf project file from archive
2. Slice it
3. Investigate layers 14 and 15

### Actual results

[image]

The support has lines in the middle of nowhere.

### Expected results

I don't expect support in the middle of nowhere

### Project file & Debug log uploads

[debug_Wed_Feb_26_10_51_50_326819.log](https://github.com/user-attachments/files/18981549/debug_Wed_Feb_26_10_51_50_326819.log)

### Checklist of files to include

- [x] Log file
- [x] Project file

miceanite, 2025-02-27:
I have this problem in 2.2.0, as well. Looks like you are probably using "Hollow" for your support base pattern. The only fix I have found is to choose something besides "Hollow."

Felixoid, 2025-05-29:
I still think it's a bug