OrcaSlicer/OrcaSlicer #8579: Crashing when using the cut tool

myxxmikeyxx, 2025-02-26:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

macOS 14.2.1

### Additional system information

[image]

### Printer

Elagoo Neptune 4 Max

### How to reproduce

I brought in an obj file model of some gears and attachment pieces.
Imported it as inches.
Scaled the project up to 240%.
Click on the object and used the cut tool.
Next I tried to align the cut (I rotated the cut plane to be vertical 90 deg)
Then I tried to move the cut plane in the X direction some to cut one piece off to rotate it.
Next i right clicked to try and rotate the camera angle to a better angle and it crashed as soon as I right clicked.
I tried it on a different model and it crashed as well. 

### Actual results

It doesn't rotate the camera. It crashes and the mac then gives me an unexpected program closure, do you want to relauch, ignore, or report issue.

### Expected results

I should be able to rotate the camera angle when using the cut plane to get a better view of where I am cutting.

### Project file & Debug log uploads

[Logs - Archive.zip](https://github.com/user-attachments/files/18985216/Logs.-.Archive.zip)

I dropped the obj file in.

[Gears.obj.zip](https://github.com/user-attachments/files/18985231/Gears.obj.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

### Anything else?

I am not sure but I don't think this happened on the last version but I could be wrong.

myxxmikeyxx, 2025-02-26:
Mac Report Error Log:

[mac-error.log](https://github.com/user-attachments/files/18985391/mac-error.log)

Noisyfox, 2025-02-27:
Please give #8598 a try

PR #8598 (2025-03-01): Cut: Fix crash caused by unexpected copy of the cut island
This copy frees the OpenGL VBOs and IBOs at the end of the loop and cause the crash.

Fix #8579
closed 2025-03-01 by SoftFever