OrcaSlicer/OrcaSlicer #8580: "total_toolchanges placeholder not correct if Prime Tower is not enabled"

ImSundee, 2025-02-26:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 Beta

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Voron 2.4 with Boxturtle

### How to reproduce

Have a multi filament print.
Disable purge tower.
have a start gcode with:  
**TOOLCHANGES=[total_toolchanges]** 

Attached 2 slices 1 with prime tower enabled  1 with it disabled.

### Actual results

TOOLCHANGES=0

### Expected results

TOOLCHANGES=394

### Project file & Debug log uploads

[MissileToad_Loop_Rev2_R-Voron 2.4 MMU-ABS+.zip](https://github.com/user-attachments/files/18985290/MissileToad_Loop_Rev2_R-Voron.2.4.MMU-ABS%2B.zip)

[toolchangevariablenowork.zip](https://github.com/user-attachments/files/18985313/toolchangevariablenowork.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

ImSundee, 2025-02-27:
Looks like it was "attempted" to be resolved in prusa slicer too: 
https://github.com/prusa3d/PrusaSlicer/issues/6832

ImSundee, 2025-05-29:
Bumping up this is still a major issue and hopefully should be a quick resolve

icu-hobby, 2025-08-04:
Same bug is described here:
https://github.com/SoftFever/OrcaSlicer/issues/10099
https://github.com/SoftFever/OrcaSlicer/issues/9737