OrcaSlicer/OrcaSlicer #8589: 2.3.0-beta - lost custom Filament Profiles after re-login

Ivan-753, 2025-02-26:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-beta

### Operating System (OS)

Windows

### OS Version

10

### Printer

inherited from greneric Klipper profile

### How to reproduce

1. Login expired and network plugin asked me to re-login
2.  Successful login
3. Create new project
4. Select printer profile
5. Select filament profile

### Actual results

No custom Filament profiles in the list

### Expected results

List of my custom Filament profiles, like it was before re-login

### Project file & Debug log uploads

Just Empty / New Project

[log.zip](https://github.com/user-attachments/files/18993150/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

### Anything else?

I reinstalled stable v2.2.0 and removed beta version. All filament profiles are back.

Bykador, 2025-02-27:
Здравствуйте, столкнулся с такой же проблемой. После обновления на 2.3.0 пропали все профили нитей. Откатился на 2.2.0 и все вернулось.

jonathanalevi, 2025-02-28:
Same, except for me it's all my printers missing. Glad rolling back might help...

luksto, 2025-03-01:
First, do **Backups**! Ether with a `*.orca_printer` file (in OrcaSlicer go to `File` -> `Export` -> `Export Preset Bundle...` select everything) or manually (by copying the config directory, see later).

I head the same problem and also my `backup.orca_printer` could not restore my manually set User-Filaments.
Luckily, the `*.orca_printer` file is an archive file (like zip) which holds the `filament`, `process` and `machine`(your printers) config directories.

My **solution** was to manually copy all filament `*.json` and `*.info` files from the extracted `backup.orca_printer` to my config directory in `~/.config/OrcaSlicer/user/default/filament/`. After restarting the OrcaSlicer the filaments are back.
For Windows this path may is under AppData o.s. and for Mac I absolutely don't know, but probably somehow like it's Linux.

If you want to try it in a Dry-run you can create a "new" temporary home directory, f.i. in `/tmp/home/.config/` and set your `HOME` environment variable to this path before the orca-slicer command. For me, it looks like this:
`HOME=/tmp/home/ orca-slicer` 
Then OrcaSlicer opens like on a fresh install, you set your region, printer usw., close it again and then copy the filament files (like described above) and start it again, same command.
This does not touch your "main" installation and setup.

I wish Orca would have a native command line option for the config path....

closed 2025-06-07 by github-actions[bot]