OrcaSlicer/OrcaSlicer #8604: Crash when slicing model; debug log stating: "Error while extract auxiliary file to file"

patschi, 2025-02-27:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-beta

### Operating System (OS)

Windows

### OS Version

Windows 11 23H2

### Additional system information

n/a

### Printer

Bambu Lab X1C

### How to reproduce

1. Start OrcaSlicer
2. Load .3mf model: https://makerworld.com/en/models/1140552-cute-elephant-for-flowers-decorative-pot-vase
3. Click "Slice plate" without changing anything else
4. Then "Generating support" occurs and takes a few seconds:
[image]
5. Then crashes to desktop

### Actual results

Crash to desktop.

### Expected results

No crash

### Project file & Debug log uploads

The debug file contains:
```
[warning]	2025-02-27 18:46:23.436571[Thread 0x00001d68]:Slic3r::NetworkAgent::get_network_function, can not find function bambu_network_get_model_id_from_desgin_id
[warning]	2025-02-27 18:46:23.437071[Thread 0x00001d68]:Slic3r::NetworkAgent::get_network_function, can not find function bambu_network_get_profile_3mf
[error]	2025-02-27 18:46:28.613506[Thread 0x00001d68]:Slic3r::NetworkAgent::connect_server error: network_agent=000001E4EA939660, ret=-2
[warning]	2025-02-27 18:46:29.010933[Thread 0x00007138]:[Orca Updater]: online version=1.10.1.9, current_version=1.10.1.9, no need to download
[error]	2025-02-27 18:46:30.533325[Thread 0x00001d68]:Error while extract auxiliary file to file
```

Crash file:
```
Exceptions:

Exception Code :c0000005 ACCESS_VIOLATION

Exception Flag :0x0 NumberParameters :2 
Param 0 :0x0 
Param 1 :0x0 
Context :000000CE9A0F6040 
ContextFlag : 0x10005f, EFlags: 0x10246 
Fault address:  0x3E87A19B 0x1:0x6A919B C:\Program Files\OrcaSlicer\OrcaSlicer.dll

Show CallStack:
```

### Checklist of files to include

- [x] Log file
- [x] Project file

fluentd, 2025-02-27:
Just came here to report the same thing. Orca was working just fine the other day. Now today I can't slice a single model. Tried multiple files, tons of models. Here is one of the errors that I get some of the time. 

[image]

Bradford1040, 2025-02-27:
Windows 10 same! Only if I open it back right after crash and reload the last option I can slice it, and print

It's a beta so can't get overly frustrated, and it's free!

Hope they get it fixed soon, as I like the new features!

Noisyfox, 2025-02-28:
> Crash file:
>
> ```
> Exceptions:
>
> Exception Code :c0000005 ACCESS_VIOLATION
>
> Exception Flag :0x0 NumberParameters :2
> Param 0 :0x0
> Param 1 :0x0
> Context :000000CE9A0F6040
> ContextFlag : 0x10005f, [...]

Could you upload the entire crash report file?

Noisyfox, 2025-02-28:
Managed to reproduce the crash, will take a look.

patschi, 2025-02-28:
I can confirm 2.3b2 does indeed fix it. No crash on the same model.

Raxcoms, 2025-03-17:
I had this happen to me a couple of times with 2.3.0-rc

`Exceptions:
Exception Code :c0000005 ACCESS_VIOLATION
Exception Flag :0x0 NumberParameters :2 
Param 0 :0x0 
Param 1 :0x5c64890 
Context :000000119164A840 
ContextFlag : 0x10005f, EFlags: 0x10282 
Fault address:  0xB06297D3 0x1:0xA587D3 C:\Program Files\OrcaSlicer\OrcaSlicer.dll

Show CallStack:
7ffcb06297d3:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffcb0622089:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffcb063b72e:BRepExtrema_SelfIntersection::`default constructor closure' [Unknown file][-1]
7ffcafeb8947:Unknown [Unknown file][-1]
7ffcafeb40c2:Unknown [Unknown file][-1]
7ffcafebf008:Unknown [Unknown file][-1]
7ffcafec66d9:Unknown [Unknown file][-1]
7ffcafec659b:Unknown [Unknown file][-1]
7ffde3c737b0:wcsrchr [Unknown file][-1]
7ffde50ee8d7:BaseThreadInitThunk [Unknown file][-1]
7ffde619bf6c:RtlUserThreadStart [Unknown file][-1]

Logical Address:
0xB06297D3 0x1:0xA587D3 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0xB0622089 0x1:0xA51089 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0xB063B72E 0x1:0xA6A72E C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0xAFEB8947 0x1:0x2E7947 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0xAFEB40C2 0x1:0x2E30C2 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0xAFEBF008 0x1:0x2EE008 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0xAFEC66D9 0x1:0x2F56D9 C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0xAFEC659B 0x1:0x2F559B C:\Program Files\OrcaSlicer\OrcaSlicer.dll
0xE3C737B0 0x1:0x27B0 C:\WINDOWS\System32\ucrtbase.dll
0xE50EE8D7 0x1:0x2D8D7 C:\WINDOWS\System32\KERNEL32.DLL
0xE619BF6C 0x1:0xBAF6C C:\WINDOWS\SYSTEM32\ntdll.dll
`
Seems to happen more often when I have a lower threshold angle set for supports

Noisyfox, 2025-03-18:
> I had this happen to me a couple of times with 2.3.0-rc
>
> ```
> Exceptions:
> Exception Code :c0000005 ACCESS_VIOLATION
> Exception Flag :0x0 NumberParameters :2
> Param 0 :0x0
> Param 1 :0x5c64890
> Context [...]

Do you have a 3mf file that could reproduce this issue? Also, how many threads does your cpu have?

Raxcoms, 2025-03-18:
I have a R7 5700X 3D (8C/16T)

3mf: (sending compacted as github doesn't seem to allow .3mf files)
[Crash.zip](https://github.com/user-attachments/files/19315709/Crash.zip)

It contains this model: 
https://www.thingiverse.com/thing:4834113

Noisyfox, 2025-03-18:
> I have a R7 5700X 3D (8C/16T)
> 
> 3mf: (sending compacted as github doesn't seem to allow .3mf files) [Crash.zip](https://github.com/user-attachments/files/19315709/Crash.zip)
> 
> It contains this model: https://www.thingiverse.com/thing:4834113

Thanks, I'm able to reproduce the crash. Will take a look soon.

Raxcoms, 2025-03-19:
Seems to be working fine on the latest nightly
Thanks!

PR #8955 (2025-03-18): Fix crash when support bottom interface layer is set to `same as top` 
Fix #8604