OrcaSlicer/OrcaSlicer #8605: Polymaker Polylite PLA keeps on syncing to Polylite Dual PLA

marvin968, 2025-02-27:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-beta

### Operating System (OS)

macOS

### OS Version

macOS 15.3.1

### Printer

Bambu Lab A1

### How to reproduce

1. Under Device > Manually selected PolyLite PLA as Filament
2. Selected color
3. Clicked Confirm
4. AMS Showing Filament:PolyLite PLA
5. Under Prepare > Filament > Clicking on Sync or Resync will show my Filament as PolyLite Dual PLA

### Actual results

[image]

[image]

### Expected results

Filament should show and apply the correct Polymaker Polylite PLA Profile

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19014175/log.zip)

[For OrcaBug.3mf.zip](https://github.com/user-attachments/files/19014185/For.OrcaBug.3mf.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

carpaij, 2025-03-26:
I have the exact same issue with PolyTerra PLA since I upgraded OrcaSlicer to 2.3.0. It syncs as PolyTerra Dual PLA

[image]

[image]

PhapandNap, 2025-04-01:
Same issue with Polly Maker PLA. The device page has it right when I set it. The model view page changes it to Dual PLA when I sync.

sbrocket, 2025-05-01:
+1, same issue with PolyTerra PLA in the AMS. It's not just the sync feature either. The drop-down list also shows "PolyTerra Dual PLA" instead of "PolyTerra PLA". The full profile name that gets selected is "PolyTerra Dual PLA @System", instead of "PolyTerra PLA @BBL X1C".

[image]

h3farooqi, 2025-07-11:
Same happens for me with PolyTerra, though note that you can get around this until they fix it by creating a custom filament (in the menu next to the AMS syncing button on the "Prepare" section) and copying the appropriate filament preset. This will create a new filament you can select for your AMS/lite just like you would with the default Polymaker filament presets, and it will sync up properly.

I would show screenshots but I currently have no functioning screenshot button...