OrcaSlicer/OrcaSlicer #8606: Segfault on opening

iplayfast, 2025-02-27:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

OrcaSlicer-01.09.05.51:

### Operating System (OS)

Linux

### OS Version

linux mint 24

### Additional system information

I had it working, and configure my voron trident wifi, (tested and it said it worked), clicked on topleft button trying to figure out how to add new printer and it crashed.
I then reopened it, and it crashed again. 
```
./OrcaSlicer_Linux_Ubuntu2404_V2.2.0.AppImage 
[2025-02-27 13:12:08.364759] [0x00007d2e34123500] [trace]   Initializing StaticPrintConfigs
add font of HarmonyOS_Sans_SC_Bold returns 1
add font of HarmonyOS_Sans_SC_Regular returns 1
add font of NanumGothic-Regular returns 1
add font of NanumGothic-Bold returns 1

(orca-slicer:1779093): Gtk-CRITICAL **: 13:12:08.739: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:1779093): Gtk-CRITICAL **: 13:12:08.739: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:1779093): Gtk-CRITICAL **: 13:12:08.739: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:1779093): Gtk-CRITICAL **: 13:12:08.740: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:1779093): Gtk-CRITICAL **: 13:12:08.740: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:1779093): Gtk-CRITICAL **: 13:12:08.740: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:1779093): Gtk-CRITICAL **: 13:12:08.740: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:1779093): Gtk-CRITICAL **: 13:12:08.741: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:1779093): Gtk-CRITICAL **: 13:12:08.742: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:1779093): Gtk-CRITICAL **: 13:12:08.742: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:1779093): Gtk-CRITICAL **: 13:12:08.743: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:1779093): Gtk-CRITICAL **: 13:12:08.743: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:1779093): Gtk-CRITICAL **: 13:12:08.744: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:1779093): Gtk-CRITICAL **: 13:12:08.744: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:1779093): Gtk-CRITICAL **: 13:12:08.744: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:1779093): Gtk-CRITICAL **: 13:12:08.744: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:1779093): Gtk-CRITICAL **: 13:12:08.744: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:1779093): Gtk-CRITICAL **: 13:12:08.744: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:1779093): Gtk-CRITICAL **: 13:12:08.911: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:1779093): Gtk-CRITICAL **: 13:12:08.914: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:1779093): Gtk-CRITICAL **: 13:12:08.914: gtk_widget_set_size_request: assertion 'width >= -1' failed

[... 246 lines cut ...]

### Printer

voron trident (klipper)

### How to reproduce

Start the application.

### Actual results

See above.

### Expected results

not crash

### Project file & Debug log uploads

I didn't get that far.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2025-02-28:
See #7210

iplayfast, 2025-03-01:
Yes that's the same issue.

closed 2025-03-01 by iplayfast