OrcaSlicer/OrcaSlicer #8609: 2.3-beta - elegoo orangestorm giga profile needs to be fixed.

GitZapf, 2025-02-28:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

On September 30th of 2024, a pull request was made to add the elegoo orangestorm giga: https://github.com/SoftFever/OrcaSlicer/pull/6977

This profile was supposed to be based off the profiles elegoo had developed for their fork of orcaslicer at the time: https://github.com/SoftFever/OrcaSlicer/issues/6256#issuecomment-2362430471

Shortly thereafter, it was pointed out that there were both organizational and just plain slicing issues with said pull request: https://github.com/SoftFever/OrcaSlicer/pull/6977#issuecomment-2443154075

I downloaded the standalone beta of 2.3, as well as elegoo's recently released fork of orcaslicer (released with their CC printer, that appears to have the same OSG profile as the one shared in september 2024) and I can confirm that orcaslicer's OSG profile still has these issues.

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

Pull elegoo's osg printer profiles from the current elegooslicer fork (or maybe just all of their official print profiles?) into orcaslicer 2.3

tlhintoq, 2025-03-02:
thanks for the reference to my earlier comment.
I've been manually replacing the directory on my own machine with every Orca update.  Kinda annoying but hey, what can ya do?

by the way - I updated the Elegoo machines profile folder with the new Centauri models if anyone cares to use them.
https://github.com/SoftFever/OrcaSlicer/issues/6256#issuecomment-2692562097

GitZapf, 2025-08-22:
https://github.com/ELEGOO-3D/ElegooSlicer/tree/main/resources/profiles/Elegoo - up to date elegoo profiles, including an orcaslicer profile with the fixed default speeds.