OrcaSlicer/OrcaSlicer #8616: [REGRESSION FOM Bambulab Studio] Unable to create manual support

LightDestory, 2025-02-28:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 Beta

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambulab X1C

### How to reproduce

1. Go to https://makerworld.com/it/models/16606-pastamatic-filament-spool-winder-for-bambu-x1c-p1p#profileId-26699 and open the latest design from author;
2. Delete every object except for "2x_crossbar_Pastamatic_Filament_Winder_by_GekoPrime_2023-09-30"
3. Draw a manual support like this one: 
[image]
4.  Slice

### Actual results

No support has been created.

### Expected results

The creation of a support. I performed the same steps with Bambulab Studio 1.10.2 and everything works as excepted.

### Project file & Debug log uploads

[No_Support_Created.zip](https://github.com/user-attachments/files/19029485/No_Support_Created.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Noisyfox, 2025-03-01:
Vertical support painting is not supported until 2.3 beta2

LightDestory, 2025-03-01:
@Noisyfox My bad then. I am sorry for opening this issue xD

Thanks for the info!

closed 2025-03-01 by LightDestory