OrcaSlicer/OrcaSlicer #8619: Only one wall for top surface overrides 0 wall setting

madshaven, 2025-02-28:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

macOS 15.3.1

### Printer

Creality K1

### How to reproduce

1. Add primitive --> Cube (or any other model)
2. Set wall to 0 on a default profile
3. Slice
4. See that the model has walls!

Untick of "Only one wall on top surfaces" needed to remove walls.

### Actual results

Before untick:
[image]

After untick:
[image]

### Expected results

"Only one wall on top surfaces" should not override "0 walls" setting.

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2025-03-01:
[image]

I'm not able to reproduce the issue, might be something related to the profile. Could you provide a 3mf file to make sure we are on the same settings?

madshaven, 2025-03-02:
Like this?
[Cube_walls.3mf.zip](https://github.com/user-attachments/files/19044475/Cube_walls.3mf.zip)

Noisyfox, 2025-03-03:
> Like this? [Cube_walls.3mf.zip](https://github.com/user-attachments/files/19044475/Cube_walls.3mf.zip)

Exactly, I'm able to reproduce this issue now with your project. Will take a look soon.

[image]

Noisyfox, 2025-03-03:
Issue was caused by #6236

closed 2025-03-03 by SoftFever

madshaven, 2025-03-11:
This might be related. Below is one layer, on a slope, with 0 top/bottom shells. The outer and inner walls are split in two. Removing "1 wall for top surface" fix this.

[image]

[wallsplit.3mf.zip](https://github.com/user-attachments/files/19184509/wallsplit.3mf.zip)