OrcaSlicer/OrcaSlicer #8620: AMS filament list missing "Generic PLA"

wwebers, 2025-02-28:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-beta

### Operating System (OS)

Linux

### OS Version

Ubuntu 24.04

### Printer

Bambu X1C (X1Plus)

### How to reproduce

1. Insert a custom filament into a AMS slot
2. Open Device and select the printer
3. Select the newly filled AMS slot containing the custom filament
4. Open the drop down list of filaments and search for the list with generic filament types

### Actual results

List containing "Generic PHA" bu not "Generic PLA"

[image]

### Expected results

List containing "Generic PLA"

[image]

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

wwebers, 2025-02-28:
Without "Generic PLA" selectable its hard to use filament profile based on that generic profile, like e.g. those from AddNorth.

Hamsterberg, 2025-03-01:
Schau mal z.B. bitte hier:

https://github.com/SoftFever/OrcaSlicer/issues/8545

und hier:

https://github.com/SoftFever/OrcaSlicer/issues/8509

Please have a look e.g. here:

https://github.com/SoftFever/OrcaSlicer/issues/8545

and here:

https://github.com/SoftFever/OrcaSlicer/issues/8509

Bwee's last post is the solution.

wwebers, 2025-03-13:
Thanks, the beta2 fixes this.

closed 2025-03-13 by wwebers