OrcaSlicer/OrcaSlicer #8622: Crash when slicing (multimaterial?)

oullah, 2025-02-28:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-beta2 Build 51916ff

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

AMD Ryzen 7 7700X 8-Core Processor, 4501 Mhz, 8 Core(s), 16 Logical Processor(s)
32GB RAM
NVIDIA TITAN RTX

### Printer

Bambu Lab P1S

### How to reproduce

1. Slice attached 3mf file

### Actual results

Crash

### Expected results

No crash

### Project file & Debug log uploads

[runner, armor.zip](https://github.com/user-attachments/files/19032387/runner.armor.zip)

[debug_Fri_Feb_28_14_28_26_36156.log](https://github.com/user-attachments/files/19032409/debug_Fri_Feb_28_14_28_26_36156.log)

[crash_Fri_Feb_28_14_12_28_0.log](https://github.com/user-attachments/files/19032410/crash_Fri_Feb_28_14_12_28_0.log)

[debug_Fri_Feb_28_14_11_54_42844.log](https://github.com/user-attachments/files/19032414/debug_Fri_Feb_28_14_11_54_42844.log)

[crash_Fri_Feb_28_14_11_46_0.log](https://github.com/user-attachments/files/19032419/crash_Fri_Feb_28_14_11_46_0.log)

[debug_Fri_Feb_28_14_11_31_11292.log](https://github.com/user-attachments/files/19032421/debug_Fri_Feb_28_14_11_31_11292.log)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

I think this is multimaterial related. I didn't have any issues when printing with one material.

xdrr, 2025-03-01:
I have a similar issue when slicing. Version 2.3.0-beta. Occurs when generating supports for a multi-material plate using PLA and PLA-supp as the support interface. Device X1C running latest official updates.

wwian, 2025-03-01:
This happened to me last night, as well.
Slicing a multi-plate model, using PETG and selecting PLA as the support/ raft interface.

manjaroblack, 2025-03-01:
This is an intermittant issue for me as well. With tree supports enabled on a model that is above a certain size it fails 90% of the time, crashing OrcaSlicer. Occasionally it will succeed, but it is rare.

Windows 11 in a VM, no GPU.
Orca Slicer 2.3.0 Beta 2
Custom Klipper (Kalico) printer

closed 2025-06-07 by github-actions[bot]