OrcaSlicer/OrcaSlicer #8623: Nighlty Build Install - Bamboo Network Plugin.

leninkster, 2025-02-28:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

Ubuntu 2404 - 2.3.0-beta2  Builid 51916ff

### Operating System (OS)

Linux

### OS Version

Ubuntu 24.04

### Additional system information

Intel I7 -12th Gen
64Gb RAM
NVidia RX Video

### Printer

Elegoo Neptune 4 Pro, Neptune 4 Max, Neptune 3 via Octoprint

### How to reproduce

Run appimage
Installer says Bambu Network Plugin is 100% installed but status bar is only at 50%.  Leave for 2-3 minutes ad status bar moves to 100%.

Unknown if this would cause further issues if a user presses close while the status bar is at 50%.  

Likelyc a cosmetic issue.

### Actual results

[image]

### Expected results

Close button should become enable until the Bambu Network Installer is 100% complete

### Project file & Debug log uploads

Project file is irrelevant.  This is happening on first run of Appimage

### Checklist of files to include

- [ ] Log file
- [ ] Project file

darco415, 2025-03-01:
ive always closed it once it says 100%

Hamsterberg, 2025-03-01:
Diese Unstimmigkeit zwischen dem Fortschrittsbalken und der Zahl war schon in der Version 2.2.0 vorhanden.

This discrepancy between the progress bar and the number was already present in version 2.2.0.

Bmelani, 2025-03-09:
I just noticed this as well. Works fine but is a visual bug of the loading bar

closed 2025-06-15 by github-actions[bot]