OrcaSlicer/OrcaSlicer #8632: Crash transfer value on Compare presets

petitxef, 2025-03-01:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 beta Build dc49116

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Ender 3

### How to reproduce

1. Compare two different presets
2. Tick "Transfer values from left to right"
3. Click "Transfer"

### Actual results

Crash and do not transfer the data

### Expected results

not crash and transfer the data

### Project file & Debug log uploads

[debug_Sat_Mar_01_21_04_46_13792.log.zip](https://github.com/user-attachments/files/19040592/debug_Sat_Mar_01_21_04_46_13792.log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

manjaroblack, 2025-03-02:
I am having this issue as well.

Version 2.3.0 Beta 2

5KilosOfCheese, 2025-03-03:
Same here OrcaSlicer_Windows_V2.3.0-beta2_portable from 23/02/2025

Melanie-Hesse, 2025-03-07:
Same here. Orca 2.3.0 beta 2

PR #8744 (2025-03-07): Fix preset transfer crash
The issue is caused by #7573, the order of the combox no longer matches the order of enum `Preset::Type`, which breaks `DiffPresetDialog::get_left_preset_name` and `DiffPresetDialog::get_right_preset_name`, which rely on the combox order:
https://github.com/SoftFever/OrcaSlicer/blob/36a5a14507db6cdecbfc9b0d2768f06b18123f30/src/slic3r/GUI/UnsavedChangesDialog.cpp#L2384-L2394

This PR fixes this by using `std::find_if` instead so it no longer relies on the enum order.

Fix #7991
Fix #8632
Fix #8674
Fix #8678
Fix #8715
Fix #8732
closed 2025-03-07 by SoftFever