OrcaSlicer/OrcaSlicer #8637: Orca closes when exporting preset bundle

Stolowma, 2025-03-01:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 DEV

### Operating System (OS)

Windows

### OS Version

Win 11 22631.4890

### Additional system information

AMD Ryzen 3 3200U with Radeon Vega Mobile Gfx     2.60 GHz
20.0 GB RAM

### Printer

TwoTrees SP5 Klipper

### How to reproduce

1. Go to file
2. Go to export
3. Click  on export
4. Click on export preset bundle

### Actual results

After a few seconds delay Orca closes. No dialog box or anything else

### Expected results

I assume a dialog box as to where to save the file or something?

### Project file & Debug log uploads

[crash_Sat_Mar_01_14_08_06_0.log](https://github.com/user-attachments/files/19040896/crash_Sat_Mar_01_14_08_06_0.log)

[3M_Filter_Assembly_v1.zip](https://github.com/user-attachments/files/19040901/3M_Filter_Assembly_v1.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Noisyfox, 2025-03-02:
Try latest beta release

Stolowma, 2025-03-02:
I went back to 2.2.0 and issue went away so I'll wait for the next full release. Thanks for the help and all your work.

closed 2025-03-27 by Stolowma