OrcaSlicer/OrcaSlicer #8641: Option to automatically connect to LAN printer(s)

daktah, 2025-03-02:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

It would be nice to have an option to have Orca automatically connect to your LAN printers when they are detected on the network, similar to how it works when using a Bambu printer via the cloud, so you don't have to switch to the device tab and connect manually each time you boot up Orca.
I feel this would be a nice QOL improvement, especially for those of us who have migrated off of the Bambu cloud for obvious reasons.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Have an option either in the preferences menu or the device tab to automatically connect to your bound printers when they are detected after Orca starts. Also the ability to set a default/main printer if appropriate (I only have 1 printer so I am unaware of any nuances of running multiple at once over LAN).

Noisyfox, 2025-03-02:
It's already added in 2.3 beta.

daktah, 2025-03-04:
I should have known you guys would be on top of it!

closed 2025-03-04 by daktah