OrcaSlicer/OrcaSlicer #8642: Missing Library in the appimage = libwebkit2gtk

yapvoonyee, 2025-03-02:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

Nixos using KDE Latest

### Additional system information

Operating System: NixOS 24.11
KDE Plasma Version: 6.2.4
KDE Frameworks Version: 6.8.0
Qt Version: 6.8.0
Kernel Version: 6.6.63 (64-bit)
Graphics Platform: Wayland
Processors: 4 × Intel® Core™ i5-2400S CPU @ 2.50GHz
Memory: 3.8 GiB of RAM
Graphics Processor: AMD TURKS
Manufacturer: Apple Inc.
Product Name: iMac12,1
System Version: 1.0

### Printer

Solvol SV06

### How to reproduce

$appimage-run OrcaSlicer_Linux_V2.2.0.AppImage 

### Actual results

OrcaSlicer_Linux_V2.2.0.AppImage installed in /home/yapvoonyee/.cache/appimage-run/deea80dcf5d3aeb384d25f3389102d990d3b801141fb5671dd2e8986638fad9f
/home/yapvoonyee/.cache/appimage-run/deea80dcf5d3aeb384d25f3389102d990d3b801141fb5671dd2e8986638fad9f/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory

### Expected results

it should find the library and run

### Project file & Debug log uploads

cat debug_Thu_Jun_20_11_00_17_12996.log.0 
[warning]       2024-06-20 11:00:18.064245[Thread 0x00007fed60343140]:Failed to load to hints.cereal. File does not exists. /home/yapvoonyee/.config/OrcaSlicer/user/hints.cereal
[error] 2024-06-20 11:00:35.846027[Thread 0x00007fed60343140]:calc_exclude_triangles:Unable to create exclude triangles
[error] 2024-06-20 11:00:47.731572[Thread 0x00007fed60343140]:calc_exclude_triangles:Unable to create exclude triangles
[error] 2024-06-20 11:01:32.588709[Thread 0x00007fed60343140]:calc_exclude_triangles:Unable to create exclude triangles
[warning]       2024-06-20 11:01:32.608138[Thread 0x00007fed60343140]:get_version, get_version not supported,return 00.00.00.00!
[error] 2024-06-20 11:02:14.750279[Thread 0x00007fed60343140]:calc_exclude_triangles:Unable to create exclude triangles

### Checklist of files to include

- [x] Log file
- [x] Project file

Flammable-Duck, 2025-03-02:
I'm having this issue too, I'm glad It's not just me lmao

Noisyfox, 2025-03-02:
Run the ubuntu 24.04 version instead.

DannyMac, 2025-03-02:
I'm running TuxedoOS (Ubuntu-based) and getting the exact same error.

yapvoonyee, 2025-03-03:
The ubuntu one has the same problem for me.

imserhii, 2025-03-03:
fedora, gnome, nvidia 4070 (prop) - same

closed 2025-06-09 by github-actions[bot]