OrcaSlicer/OrcaSlicer #8644: Cura Features That would be nice Functional Wise for orca

unaww, 2025-03-02:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I dont mean to be comparing but i feel alot of people might enjoy using this for some extra strength 
the main one is line multipler
secoundary is infill polygons and infill lines (you can see which ones are selected)

[image]

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

If possible work on adding the line multiplyer in 
later if possible the extra infill wall count 

im willing to help how i can even if its beta testing to see if it works right 

### Describe alternatives you've considered

1 Cura
2 regret 
3 open orca boots in 1 secound

### Additional context

<3

WolfesBlueBox, 2025-03-03:
in theory, this may also reduce machine wear and rapid vibrations...

unaww, 2025-03-07:
>.> this is true for normal infills i have a feeling gyroid with this setting would lossen some bolts pretty quick..

> in theory, this may also reduce machine wear and rapid vibrations...

im just thinking more infill lines can increase infill strength without making it extra dense.

RF47, 2025-05-13:
In case anyone is interested.
https://github.com/SoftFever/OrcaSlicer/pull/9432

unaww, 2025-05-14:
Oh sweet lord thank you so much extra strong prints! 
<3

unaww, 2025-05-14:
now to figure out how to auctully use it in the slicer more research for mee.

ianalexis, 2025-06-30:
Merged in Main with #9432 
And you can use it in the last [Nightly Builds](https://github.com/SoftFever/OrcaSlicer/releases/tag/nightly-builds)
@unaww You can close this issue now.

closed 2025-06-30 by unaww