OrcaSlicer/OrcaSlicer #8654: Preview Pane Layer Slider Sensitivity

kornface13, 2025-03-02:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Yes.  When trying to add a filament change at a certain layer, the layer slider in the preview tab is overly sensitive On OSX you two finger click/tap on the trackpad for a right click.  When you do this in the Preview pane on the layer slider to add a filament change, most of the time it will change layers (almost always moves exactly 2 layers down no matter how I click).  You cannot edit it, so you have to delete the color change and try again.  This is frustrating when you need to add a bunch of color changes.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It would also be nice if there was just a button to "Add filament change" and you put in the layer number without having to move sliders at all. After moving sliders, you just have to move them right back to where they were to actually see the color change anyways, so this could be avaoided completely.

If that isn't feasible, maybe make the color change button editable so that you can either drag it up/down to different layer heights, or just "edit layer" and manually enter an exact layer.  

[image]

### Describe alternatives you've considered

1. I've tried clicking the trackpad every way imaginable, but I can't find a way that doesn't result in a layer shift.  
2. External mouse, but I don't even own a USB-C mouse so this is far from ideal.

ShaneDelmore, 2025-03-05:
Thanks for making this issue.  I kept trying to remember to make an issue regarding this exact behavior for ages now and kept getting distracted and forgetting about it.

jtenniswood, 2025-09-13:
I just found this issue too, it's like they have used two fingers for shuttling, but that makes it impossible to accurately right click!

jtenniswood, 2025-09-13:
I've just tried this with the macic mouse and the same issue still exists, it's impossible to right click without moving the layer. Completely infuriating to use.