OrcaSlicer/OrcaSlicer #8656: Printer panel shows on Device tab when some BL AMS are selected

louhy, 2025-03-02:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Linux

### OS Version

Linux Mint 21.1

### Printer

Bambu Lab X1C

### How to reproduce

1. Go to Prepare tab
3. Go to Device tab
4. Connect to printer (Printer panel shows)
5. Select each AMS on the right "Control" panel to make panel hide/show

### Actual results

When _some_ AMS are selected, the Device panel (which is normally only displayed on Prepare or Preview tabs) pops up on the left side of the window, as shown. Which AMS might change between sessions, but at this moment selecting the first or last AMS makes the panel hide, and selecting the 2nd or 3rd AMS makes the panel appear.

[image]

### Expected results

"Printer" panel should never show when the Device tab is active

### Project file & Debug log uploads

Project file unnecessary. This happens for all project files, including a brand new empty project file.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

louhy, 2025-03-02:
I seem to have found a workaround. This only seems to happen when the Printer panel is floating (undocked). I was able to drag the panel from the top to the left side of the window, and get it to snap/dock on the Prepare tab. It looks like this prevents the issue. Should I leave this open since it's still a bug with the panel in floating mode, or close?

Noisyfox, 2025-03-03:
> Which AMS might change between sessions, but at this moment selecting the first or last AMS makes the panel hide, and selecting the 2nd or 3rd AMS makes the panel appear.

What does "selecting AMS" mean?

louhy, 2025-03-04:
I'll throw up a screenshot when I get a chance (printer needs to be on and connected to show that). Basically on the device tab the right side has a panel which shows 4 'buttons' if you have 4 Bambu Lab AMS. Probably not the most common setup out there, so good question.

louhy, 2025-03-16:
@Noisyfox I added a screenshot to the description at the end, to hopefully make that clearer. The workaround to dock the floating window has been working fine though, TBH. So this is a fairly low priority bug in my view.

Noisyfox, 2025-03-17:
Please give PR #8934 a try and see if it solves the problem

louhy, 2025-03-22:
@Noisyfox  Hey, how'd you know I'd be able to figure out how to build that PR locally and test it? 
I like the repo folder 'bugfox'.  :joy:

The fix looks like it works. Very cool!

PR #8934 (2025-04-05): Make sure sidebar is not triggered by child notebook tab changes 
such as AMS group selection.
Fix #8656 

Could not reproduce this myself, need verification from original poster, and anyone who has the same setup.