OrcaSlicer/OrcaSlicer #8658: top shell layer stuck to concentric (v2.3.0-beta2)

JayceeB1, 2025-03-02:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

v2.3.0-beta2

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

BLV MNG Cube

### How to reproduce

New project, tried to change top surface pattern.
All selected patterns produces concentric layers 

### Actual results

[image]

### Expected results

This should change setting...

### Project file & Debug log uploads

[3DPNFilHoldRoller.zip](https://github.com/user-attachments/files/19044510/3DPNFilHoldRoller.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

Thanks for the awesome work :)

Noisyfox, 2025-03-03:
Because that's your ironing pattern.

JayceeB1, 2025-03-03:
Yep... sorry for this, I didn't noticed earlier!

closed 2025-03-03 by JayceeB1