OrcaSlicer/OrcaSlicer #8661: Prime Tower Brim Generates Moves Beyond Plate Boundaries

achulzhanov, 2025-03-03:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

Intel Core i7-13700K, 32GB DDR5-6000, Nvidia RTX 4070 Super

### Printer

Voron 2.4 350mm, Box Turtle MMU

### How to reproduce

1. Disable 'purge into prime tower' in printer settings to enable additional 'prime volume' setting in process (global) > multimaterial (purging into prime tower enabled does not present this issue).
2. Set prime volume to a small amount, i.e. 1-5 mm^3 (15 mm^3 or above does not present this issue).
3. Set prime tower brim width to 4mm or greater (brim widths 3mm and under do not present this issue).
4. Prepare a plate where the print has a tool change on the first layer, either with different materials for separate parts or a multi color/material single part.
5. Generate the gcode and note the error 'A G-code path goes beyond plate boundaries' with print lines seemingly randomly placed on the print bed during the brim printing operation.

### Actual results

[image]

As pointed out in the screenshot, G1 commands to move the toolhead near the corner of the bed appear before the M73 print progress commands during the outer wall of the prime tower brim. The second G1 movement in this issue goes even further and commands moves outside of the print bed before returning to the expected position for the prime tower brim.

### Expected results

This error and G1 commands should not appear - prime volume (which I don't know why is made an additional required setting when purging into prime tower is disabled in the first place) and brim width should not produce these odd, random lines on the print bed and command moves outside of the printable area.

### Project file & Debug log uploads

[project.zip](https://github.com/user-attachments/files/19046432/project.zip)

[log.zip](https://github.com/user-attachments/files/19046439/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

I have not done enough projects yet to know how repeatable this issue is, but I have seen this same behavior with a different STL once before. It seems to be for any slicing where there is a tool change on the first layer, as it does not do this on single color/material prints or multi material prints where there is no tool change on the first layer when the prime tower brim is printed.

mattiras, 2025-03-25:
Came to see if reported. I got this behavior with 2.3.0 official release. Please fix.

mattiras, 2025-03-25:
Actually, I don't know if my bug is the same. I have a tool changer setup and "Purge in prime tower" is enabled.

Decided to open a separate report #9078

drake7707, 2025-04-26:
[image] Just ran into the same thing. Changing the purge tower width (from 60mm to 50mm) seems to fix it

closed 2025-08-02 by github-actions[bot]