OrcaSlicer/OrcaSlicer #8664: YOLO flow rate calibration flow rates do not match

manjaroblack, 2025-03-03:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 Beta 2

### Operating System (OS)

Windows

### OS Version

11

### Printer

Custom Klipper

### How to reproduce

1. Select Calibration
2. Select Flow Rate
3. Select a YOLO calibration
4. Select one of the objects
5. View the object process settings
6. Under Quality observe the flow rate and match it to the flow rate delta listed on the object

### Actual results

Selecting the .01 object yields a 1.0088 flow rate, 0.005 is 1.0044, etc.

### Expected results

Selecting the .01 object yields 1.01 flow rate, 0.005 is 1.005, etc

### Project file & Debug log uploads

.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

igiannakas, 2025-03-03:
It multiplies with your selected filament flow rate to give you the resulting printed object rate. Can you check that your filament rate is either 1.0 or the math between filament rate * object rate = the needed object filament flow rate

manjaroblack, 2025-03-03:
The process flow rate is a multiplier of the filament flow rate and is not dependent on the filament flow rate. Selecting YOLO, the objects process flow rate setting does not match what’s stated on the object. You can observe this as stated in my original bug report (the object that has .01 on it has a flow rate of 1.0088 but should be 1.01). You can see this working correctly if you look at Pass 1 and Pass 2 objects. Their process flow rates match the physical object (an object with 10 on it gives a process flow rate of 1.1)

igiannakas, 2025-03-03:
The two pass calibration is relative.

Yolo is absolute.

[image]

In the example above, the 0.01 means an effective flow rate of 0.995 -> 1.0102 (object flow ratio)*0.985 (filament flow ratio) =0.995 (effective object flow ratio).

manjaroblack, 2025-03-03:
I see, that makes sense now. Is there documentation on this?

closed 2025-03-03 by manjaroblack