OrcaSlicer/OrcaSlicer #8683: Need a way to block supports on engraved text

tlhintoq, 2025-03-04:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Because engraved text doesn't yet have surfaces there's no way to block supports.
This is a huge amount of waste for one, tiny speck.

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

If a wide area has been painted with support blocking, don't put supports there.

discip, 2025-03-04:
Confirmed.
This definitely worked previously, as I printed objects with text on slanted surfaces, while the supports for the text could be blocked using **`Support Painting`**.

It seems to have been damaged along the way.

@Noisyfox
May I ask for your expertise, Sir?

Noisyfox, 2025-03-04:
(The following screenshots are all captured from 2.2.0 release)
I don't think this ever worked, unless you bake the negative part into the original model:

[image]
[image]

discip, 2025-03-04:
Unfortunately, I can't test it myself, as the build has expired, but I printed a file around the time the following commit was released: https://github.com/SoftFever/OrcaSlicer/commit/b3318306046b7f953174f222f8ea063776a8f895 and if I'm not mistaken, it worked with this build.

discip, 2025-03-04:
@Noisyfox
Unfortunately, I can't test it myself, as the build has expired, but I printed a file around the time the following commit was released: https://github.com/SoftFever/OrcaSlicer/commit/b3318306046b7f953174f222f8ea063776a8f895 and if I'm not mistaken, it worked with this build.

tlhintoq, 2025-03-20:
FYI:  Engraved text that has been support blocked... Still creating some supports in Orca  2.3.0 Stable
I know the painting only takes place on faces you 'see' in the tilt of the application view.
I have painted from top, sides, bottom, angles... just sprayed blocking from every angle and can't manage to block it.

[image]

Noisyfox, 2025-03-20:
> FYI: Engraved text that has been support blocked... Still creating some supports in Orca 2.3.0 Stable I know the painting only takes place on faces you 'see' in the tilt of the application view. I [...]

Maybe try using support blocker modifier?

tlhintoq, 2025-03-21:
Yeah.  I guess so.  Thanks

closed 2025-03-21 by tlhintoq