OrcaSlicer/OrcaSlicer #8685: 1st Layer Of Circular Bridging

Snargola, 2025-03-04:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-beta2

### Operating System (OS)

Windows

### OS Version

Windows 10 Pro

### Printer

Bambu Lad A1

### How to reproduce

1) Load the model linked below.
2) Slice at a .16 layer height.
3) In the Preview pane, scroll to layer 69.

Pictured is layer 68, showing nothing under the floor it is going to start building on layer 69. 

[image]
[image]

### Actual results

The slicer is trying to print a circle over open air. Obviously, this fails. I could add supports but I'd never be able to get them out.

### Expected results

I had thought it would bridge, then build up layers on top.

### Project file & Debug log uploads

[Mason_Jar_Greenhouse.zip](https://github.com/user-attachments/files/19062848/Mason_Jar_Greenhouse.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

This is the link to the original .stl file.

https://www.thingiverse.com/thing:2322180

discip, 2025-03-04:
Just untick **`Extra perimeters on overhangs`**:

[image]

discip, 2025-03-04:
@Snargola
Just untick **`Extra perimeters on overhangs`**:

[image]

ShaneDelmore, 2025-03-05:
@discip I thought you were going to say use sacrificial bridges so was surprised to see disable extra perimiters on overhangs.  Why does that help?

discip, 2025-03-05:
I think it should work better, because it doesn't start printing in mid air.

before:
[image]

after:
[image]

closed 2025-06-11 by github-actions[bot]