OrcaSlicer/OrcaSlicer #8687: Brim Painting Not Actually Working

Robot-Coder2010, 2025-03-04:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 Beta

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Neptune 4 Max

### How to reproduce

Go to paint brim
Either auto paint brim or paint them yourself
Slice

### Actual results

When I slice, the brims do not show up in the preview and from my knowledge, they do not print either.

### Expected results

I think that the brims should be seen in the preview

### Project file & Debug log uploads

[Project 2.zip.zip](https://github.com/user-attachments/files/19063135/Project.2.zip.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Robot-Coder2010, 2025-03-04:
Yes I did.

Noisyfox, 2025-03-05:
Could you upload a 3mf that have painted brim?

Noisyfox, 2025-03-05:
Ah indeed it's a bug. As a workaround, you could set the global brim width to any value other than 0, then the painted brims should be working then.

PR #8700 (2025-03-06): Fix issue that painted brim is not generated if brim width is set to 0
Fix #8687
[image]