OrcaSlicer/OrcaSlicer #8691: outer/inner wall printing order is messed up in 2.3.0 Beta 2

Weeris102, 2025-03-04:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 Beta-2

### Operating System (OS)

Windows

### OS Version

11

### Printer

All

### How to reproduce

1. activate outer/inner wall printing order
2. activate classic wall generation
3. slice an object with a pocket near the outer wall

### Actual results

Although outer/inner wall printing order is selected, it doesn't really printing the whole outer wall first. It leaves out the outer wall in front of a pocket, generates inner walls and then it generates the last pieces of outer walls in front of pockets near the outer wall, which would results in ugly seams and unstable walls in front of (magnet) pockets.

### Expected results

it generates the whole outer wall first, like in version 2.2.0

### Project file & Debug log uploads

[outerinner wall problem.zip](https://github.com/user-attachments/files/19069914/outerinner.wall.problem.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

https://github.com/user-attachments/assets/ec520b85-7bc2-434a-8071-72f34b2f48dc

igiannakas, 2025-03-07:
Try using inner outer inner. Outer inner is a worse version of IOI.

With IOI this issue does not occur in your project file.

Outer inner is kept for legacy purposes but for all practical use it has been superseded by IOI.

Weeris102, 2025-03-07:
> Try using inner outer inner. Outer inner is a worse version of IOI.
> 
> With IOI this issue does not occur in your project file.
> 
> Outer inner is kept for legacy purposes but for all practical use it has been superseded by IOI.

weird, I have the same issues with IOI.

https://github.com/user-attachments/assets/c836554e-4406-4df7-b8f9-ad7ce15fbaac

igiannakas, 2025-03-07:
Can you try with the latest nightly release please?

igiannakas, 2025-03-07:
Also look from layer 2 onwards. First layer is always printed outer inner I think.

Finally do you have 3 walls enabled?

Weeris102, 2025-03-07:
Just tested in the latest nightly release and it's still the same, regardless which layer, how many walls, which wall order etc.

igiannakas, 2025-03-07:
Could you post please a single plate project file with the part that is causing this issue?

Weeris102, 2025-03-07:
[outerinner wall problem.zip](https://github.com/user-attachments/files/19131580/outerinner.wall.problem.zip)

closed 2025-06-13 by github-actions[bot]