OrcaSlicer/OrcaSlicer #8697: Full Screen Issue

samiresa, 2025-03-04:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-Beta2

### Operating System (OS)

Windows

### OS Version

Os Windows 11 Pro Version 24H2 OS Build 26100.3037

### Additional system information

Processor	13th Gen Intel(R) Core(TM) i5-13600K   3.50 GHz
Installed RAM	32.0 GB (31.8 GB usable)
System type	64-bit operating system, x64-based processor
Motherboard  Z790 Gaming Pro Wifi MS-7D93
GPU                 AMD Radeon RX6650 XT
GPU Driver       24.12.1
Screen Resolution 1920x1080

### Printer

Ender 3V2 

### How to reproduce

Put Orca Slicer in Full Screen Mode using upper right hand side

[image]

### Actual results

There is a small bar of the background behind it, it appears that Orca Slicer is not actually in "full screen" even though it appears to be.  If there is movement behind Orca Slicer, I can see it active behind Orca Slicer (Ie a game or video playing behind.)

[image]

### Expected results

Orca Slicer should fill up my entire window with no space at the bottom just like this browser does.  Both of those programs are in full screen mode.

[image]
[image]

### Project file & Debug log uploads

[logs.zip](https://github.com/user-attachments/files/19076749/logs.zip)

[Cube.zip](https://github.com/user-attachments/files/19076753/Cube.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

This isn't a problem with the slicing; it appears to be all graphic-based.

Noisyfox, 2025-03-05:
Because you have taskbar auto hidden. Without that tiny empty space you'll never able to get the taskbar shown once you maximized the window.

samiresa, 2025-03-05:
Gotcha!  Thank you, sorry I looked for full screen bugs and didn't find one, didn't think to worry about the taskbar.  Question for my own understanding, why does a hidden task bar work on other programs? (ie browsers and the like)

Noisyfox, 2025-03-05:
> Question for my own understanding, why does a hidden task bar work on other programs? (ie browsers and the like)

Because we didn't use windows' own title bar. Orca's main windows is actually a borderless window that we rendered the title bar ourself, and Windows somehow just don't allow showing the taskbar if you have borderless windows truly maximized.

samiresa, 2025-03-05:
Thank you so much for taking the time to explain that.  It makes perfect sense, no reason to reinvent the wheel if it already works.  I appreciate what you guys have created

closed 2025-06-11 by github-actions[bot]