OrcaSlicer/OrcaSlicer #8702: Docker build fails with "Unable to locate package libwebkit2gtk-4.0-dev"

mobilinkd, 2025-03-05:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.4.0-dev 2025-03-04

### Operating System (OS)

Linux

### OS Version

Fedora 41

### Printer

None

### How to reproduce

1. Run `./DockerBuild.sh`

### Actual results

```
STEP 5/21: RUN apt-get update && apt-get install  -y     autoconf     build-essential     cmake     curl     eglexternalplatform-dev     extra-cmake-modules     file     git     gstreamer1.0-plugins-bad     gstreamer1.0-libav     libcairo2-dev     libcurl4-openssl-dev     libdbus-1-dev     libglew-dev     libglu1-mesa-dev     libglu1-mesa-dev     libgstreamer1.0-dev     libgstreamerd-3-dev     libgstreamer-plugins-base1.0-dev     libgstreamer-plugins-good1.0-dev     libgtk-3-dev     libgtk-3-dev     libosmesa6-dev     libsecret-1-dev     libsoup2.4-dev     libssl3     libssl-dev     libtool     libudev-dev     libwayland-dev     libwebkit2gtk-4.0-dev     libxkbcommon-dev     locales     locales-all     m4     pkgconf     sudo     wayland-protocols     wget
Hit:1 http://archive.ubuntu.com/ubuntu noble InRelease
Get:2 http://archive.ubuntu.com/ubuntu noble-updates InRelease [126 kB]
Get:3 http://security.ubuntu.com/ubuntu noble-security InRelease [126 kB]
Get:4 http://archive.ubuntu.com/ubuntu noble-backports InRelease [126 kB]
Get:5 http://archive.ubuntu.com/ubuntu noble/main Sources [1713 kB]
Get:6 http://archive.ubuntu.com/ubuntu noble/universe Sources [24.3 MB]
Get:7 http://security.ubuntu.com/ubuntu noble-security/main amd64 Packages [837 kB]
Get:8 http://archive.ubuntu.com/ubuntu noble-updates/main amd64 Packages [1160 kB]
Get:9 http://archive.ubuntu.com/ubuntu noble-updates/multiverse amd64 Packages [28.8 kB]
Get:10 http://archive.ubuntu.com/ubuntu noble-updates/universe amd64 Packages [1344 kB]
Get:11 http://archive.ubuntu.com/ubuntu noble-updates/restricted amd64 Packages [917 kB]
Get:12 http://security.ubuntu.com/ubuntu noble-security/universe amd64 Packages [1061 kB]
Get:13 http://security.ubuntu.com/ubuntu noble-security/multiverse amd64 Packages [24.2 kB]
Get:14 http://security.ubuntu.com/ubuntu noble-security/restricted amd64 Packages [880 kB]
Fetched 32.6 MB in 3s (10.4 MB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package libwebkit2gtk-4.0-dev
E: Couldn't find any package by glob 'libwebkit2gtk-4.0-dev'
E: Couldn't find any package by regex 'libwebkit2gtk-4.0-dev'
Error: building at STEP "RUN apt-get update && apt-get install  -y     autoconf     build-essential     cmake     curl     eglexternalplatform-dev     extra-cmake-modules     file     git     gstreamer1.0-plugins-bad     gstreamer1.0-libav     libcairo2-dev     libcurl4-openssl-dev     libdbus-1-dev     libglew-dev     libglu1-mesa-dev     libglu1-mesa-dev     libgstreamer1.0-dev     libgstreamerd-3-dev     libgstreamer-plugins-base1.0-dev     libgstreamer-plugins-good1.0-dev     libgtk-3-dev     libgtk-3-dev     libosmesa6-dev     libsecret-1-dev     libsoup2.4-dev     libssl3     libssl-dev     libtool     libudev-dev     libwayland-dev     libwebkit2gtk-4.0-dev     libxkbcommon-dev     locales     locales-all     m4     pkgconf     sudo     wayland-protocols     wget": while running runtime: exit status 100
```

### Expected results

The docker build succeeds.

### Project file & Debug log uploads

Field cannot be empty. This is a build issue, not a project or runtime issue.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

This is a recent failure as a build as recently as 5 weeks ago succeeded. This is apparently due to a change in Ubuntu 24.04 LTS.

mobilinkd, 2025-03-05:
There are a number of similar issues related to AppImage builds.

mobilinkd, 2025-03-05:
Changing the Dockerfile to install libwebkit2gtk-4.1-dev instead seems to have worked.

jvalecillos, 2025-04-16:
@mobilinkd Your suggestion seems to be landed in the following commit: https://github.com/SoftFever/OrcaSlicer/commit/671d3d8f70ca489f28ad56334ec0822fc07066b4

closed 2025-07-23 by github-actions[bot]