OrcaSlicer/OrcaSlicer #8706: Crashing on print project send via LAN mode

Jwn5k, 2025-03-05:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-Beta2 Build 93cb042

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

CPU: Ryzen 7 7800X3D
GPU: RTX 2070 Super
RAM: 64GB DDR5 6000

### Printer

Bambu Lab X1 Carbon

### How to reproduce

The problem occurs somewhat infrequently (the worst kind of problem, I know), but happens after I slice a model, click "Print Plate", after I select any additional settings I need (any combination, bed leveling, flow calibration, etc.), and once it is done sending to my printer over LAN mode (not cloud), the print is sent to my printer successfully, as my printer does it's thing and starts the print job, but Orca Slicer crashes immediately afterwards.

### Actual results

There is nothing really to show, as it happens on just about any print job I start, very very simple ones as well, it crashes just after it finished sending the print job to my printer over LAN.

### Expected results

Orca Slicer might crash after the print job is sent over LAN mode to the printer.

### Project file & Debug log uploads

The last 3 crash logs:

[crash_Wed_Mar_05_02_11_45_0.log](https://github.com/user-attachments/files/19085152/crash_Wed_Mar_05_02_11_45_0.log)
[crash_Wed_Mar_05_03_12_50_0.log](https://github.com/user-attachments/files/19085153/crash_Wed_Mar_05_03_12_50_0.log)
[crash_Wed_Mar_05_03_12_51_1.log](https://github.com/user-attachments/files/19085151/crash_Wed_Mar_05_03_12_51_1.log)

Project File:

[Jake Badge Bumper.zip](https://github.com/user-attachments/files/19085155/Jake.Badge.Bumper.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

There shouldn't be much else to add as far as I know.

Schildkroet, 2025-03-05:
I have the same problem with my A1. But it happens very rarely (<5%)

Jwn5k, 2025-03-06:
> I have the same problem with my A1. But it happens very rarely (<5%)

for me it seems to happen around 20-30% of the time, and occasionally I would have 3 in a row at times, so it's frequent and infrequent enough to be a problem and annoying.

Arakon, 2025-03-12:
Happens for me on RC as well, and rather regularly. I have 2 Bambu printers, an A1 and a Mini, and it attempts to connect to one of them by default. If that printer is not available, but the other is, it crashes instead of allowing me to select the other from the list of printers.

jheyneman, 2025-03-12:
These happens to me very frequently with 2.2.0 and the 2.3.0-rc.  I've found that it will crash every time if I am on a wired network or different SSID than my printer.  I have to be connected to the same exact SSID for it to print, and only if it's an STL.  If I load a 3MF, it goes back down to a 50/50 chance it crashes or times out.

One thing I noticed with the 2.3.0-rc is that it kept asking me for the IP addres s and code to reconnect to the printer when I hit Print, then later asking me for the printer serial number.

RcrossF, 2025-05-01:
This did not happen to me until 2.3.0, and now happens on both my Windows 10 & 11 machines ~30% of the time. Sometimes it crashes before sending the job, when selecting printers from the dropdown.

RcrossF, 2025-05-01:
Choosing the printer from the device tab before sending the print (take a minute to connect) seems to avoid the crash

latimschneider, 2025-09-05:
> Choosing the printer from the device tab before sending the print (take a minute to connect) seems to avoid the crash

I made the same experiance today with my X1C and A1 Mini's. 
100% crashed on the A1 Mini's (10+ prints) but not on the X1C, which was the default connection today.
If the print files were small enough, the file got through before Orca Slicer (2.3.0, Build 70931e5) crashed.

mike-l-lawson, 2025-12-25:
I have the same issue sending to an X1C. I've never had it fail to send the job to the printer, but 1/3 of the time it crashes immediately, 2/3 of the time it crashes eventually. If I reopen orca to watch the print remotely, it rarely crashes. 1/100 times, if that.

I thought it was related to having Creality Print open for another printer, but today it isn't. I just updated to 2.3.1 current release.