OrcaSlicer/OrcaSlicer #8709: Proper Support for 3Dconnexion Devices (3D Mouse)

ramosglauco, 2025-03-05:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I would like to request the implementation of proper support for 3Dconnexion devices, similar to what is available in PrusaSlicer (would it be possible to do a simple port?)

Currently, using a 3D mouse in OrcaSlicer is a terrible experience — the controls are unresponsive, the movement feels unnatural, and overall usability is far below what is expected when compared to PrusaSlicer. In PrusaSlicer, navigating and manipulating 3D objects with a 3Dconnexion device works smoothly and intuitively, which is essential for efficient workflow.

As someone who frequently works with 3D models, being able to properly use my 3D mouse is critical. I strongly believe adding this feature would significantly enhance the user experience for all users who rely on professional navigation tools.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Would it be possible to do a port from PS?

discip, 2025-03-05:
@ramosglauco
Please enter **`Ctrl + M`** and you should be presented with a corresponding settings menu.

AlexLao512, 2025-03-06:
CTRL + M then setting Translate to 15, Rotation and Zoom to 5, and then swapping Y and Z axis works well for me.

phillipsnick, 2025-03-26:
> [@ramosglauco](https://github.com/ramosglauco) Please enter **`Ctrl + M`** and you should be presented with a corresponding settings menu.

Is this option available on Mac?

discip, 2025-03-27:
[image]