OrcaSlicer/OrcaSlicer #8714: Odd wall artifacts after changing the Z offset

opcow, 2025-03-05:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-beta2

### Operating System (OS)

Windows

### OS Version

Windows 11 24H2

### Additional system information

Processor: AMD Ryzen 9 5900X 12-Core Processor               3.70 GHz
Installed RAM: 32.0 GB
System type: 64-bit operating system, x64-based processor

### Printer

Centauri Carbon

### How to reproduce

1. Click to edit preset.
2. Change the Z offset from 0 to some positive or negative value.
3. Close the preset editing window.
4. Slice the plate.

### Actual results

Anomalous horizontal offsets on walls. The more the Z offset, the more pronounced the anomalies are.

### Expected results

I think that the walls should remain essentially unchanged and smooth.

### Project file & Debug log uploads

[project-files.zip](https://github.com/user-attachments/files/19094859/project-files.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

[image]
[image]
[image]
[image]
[image]

igiannakas, 2025-03-07:
Disable arc fitting.

Or best control your z offset using gcode commands like so, in your print start g code:

```
{if (curr_bed_type=="Textured PEI Plate" || curr_bed_type=="Textured Cool Plate")} 
; Gcode offset for Textured PEI plate for all material types
SET_GCODE_OFFSET Z_ADJUST=-0.045 MOVE=1
{endif}

{if (filament_type[current_extruder]=="ABS" || filament_type[current_extruder]=="ASA")}
; Reduce squish for high temp, non fiber filled materials
SET_GCODE_OFFSET Z_ADJUST=+0.030 MOVE=1 
{endif}
```

The first if block checks the selected plate type and applies a z offset

The second block checks the material printed and applies an ADDITIONAL z offset. Eg. Textured plate -> -0.045 + 0.03 =-0.015 for ABS with a textured plate.

opcow, 2025-03-07:
@igiannakas thank you.

opcow, 2025-03-07:
@igiannakas works perfectly. Thank you.