OrcaSlicer/OrcaSlicer #8715: Crash when transfer filament settings

dpregs, 2025-03-05:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0beta2

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Ender 3 v2 Neo

### How to reproduce

I go into filament compare and set the generic creality PETG profile on left and my profile on right. I try to transfer values left to right and after I hit transfer the program crashes.

### Actual results

program crash

### Expected results

filament settings should transfer from left to right.  Worked in previous versions.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19095085/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

evilDave, 2025-03-06:
Can confirm. I haven't been able to get any transfer to work, seems to be all the time.
I can provide crash log if desired.

evilDave, 2025-03-06:
Ugh, sorry I didn't search for duplicates before commenting. Looks like the earliest report is: https://github.com/SoftFever/OrcaSlicer/issues/7991

dpregs, 2025-03-06:
> Ugh, sorry I didn't search for duplicates before commenting. Looks like the earliest report is: [#7991](https://github.com/SoftFever/OrcaSlicer/issues/7991)

I did search, not sure why I didn't see this one. My apologies

PR #8744 (2025-03-07): Fix preset transfer crash
The issue is caused by #7573, the order of the combox no longer matches the order of enum `Preset::Type`, which breaks `DiffPresetDialog::get_left_preset_name` and `DiffPresetDialog::get_right_preset_name`, which rely on the combox order:
https://github.com/SoftFever/OrcaSlicer/blob/36a5a14507db6cdecbfc9b0d2768f06b18123f30/src/slic3r/GUI/UnsavedChangesDialog.cpp#L2384-L2394

This PR fixes this by using `std::find_if` instead so it no longer relies on the enum order.

Fix #7991
Fix #8632
Fix #8674
Fix #8678
Fix #8715
Fix #8732