OrcaSlicer/OrcaSlicer #8718: Preset system for print profiles

hz61p1, 2025-03-05:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Yes, the current system for managing print profiles in OrcaSlicer requires you to create separate profiles for each set of settings. This results in the need to manage a large number of profiles, making it difficult to compare them and transfer settings between them.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I propose to add the ability to create presets that override certain parameters of the current print profile. This would allow users to apply multiple presets to a single profile for different print scenarios. For example, one preset could change the acceleration settings and another could change the extruder settings. Such a system would reduce the number of profiles required and simplify the management of settings. When creating a preset, it should be possible to name the preset for easy identification.

### Describe alternatives you've considered

The alternative is to continue to use a separate profile for each setting option, but this is inefficient and results in the accumulation of a large number of profiles. You can also manually change the settings before each print, but this increases the risk of errors and is more time consuming.

### Additional context

The introduction of a preset system will improve the user experience, making settings management more flexible and efficient. This is especially useful for users who frequently change print settings from project to project.

tlhintoq, 2025-03-06:
What do you consider a *lot* of presets?  I print a range from small part the size of a USB drive all the way up to wearable armor and only need 9 presets - and two of those are for doing TPU.

Ya kinda lost me on the "separate preset for each set of settings".  Are you trying to do a profile for `Quality` separate from `speed` separate from `strength`?

I'm curious why you have so many profiles.

[image]

hz61p1, 2025-05-10:
@tlhintoq I conduct a lot of experiments, and for each printer profile with different nozzles, I need to synchronize the profiles after changing any printing parameters or filament. That's why I would like to have presets that would override the parameters. For example, I need to test different extrusion widths with different layer heights and nozzle diameters. Currently, I have to make countless switches and clicks...

tlhintoq, 2025-05-11:
That sounds like a VERY specific need for one person's job - to force it out on everyone else.
Rewrite a ton of stuff and push it on to 20k users because 5 people across the country do this very specific in-house testing?

Just a thought... I'm assuming if you're doing all this testing then you have a set of object and set of project files set up for doing it.
Don't change the settings in the program: Change the settings per object then save it.  Its a one-time set up of the job and you're done.

For example:
You make a plate of your 10 test objects - and that plate is .2mm layer height.
Object 1 - Line width = 100% • So that's good on all printers regardless of nozzle size
Object 2 - Line width = 110% • Again, good on all nozzle sizes.
{repeat with all your desired widths}
So basically you shouldn't have any profiles that are nozzle-size specific

Plate 2 is at .24 layer height.
Copy all your objects from plate 1 and paste on plate 2
Done

Plate 3 is at .28 layer height
Copy/paste/done

You see where I'm going with this right?  Set up smarter test jobs with smarter parameters that auto-adjust and you shouldn't have to be doing a bunch of changes every time you want to test some no material or whatever

coryrc, 2025-05-13:
Essentially a dupe of #8261