OrcaSlicer/OrcaSlicer #8719: Outer wall not detected as bridge

pm0u, 2025-03-05:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 beta2

### Operating System (OS)

Linux

### OS Version

Manjaro, up to date

### Printer

Qidi X Plus 4

### How to reproduce

See attached 3mf

### Actual results

Line is printed as an outer wall with no anchoring overlap or slowdown, leading to a failure.

[image]

### Expected results

Line would be detected as bridge and anchored with overlap on both ends and slowed appropriately

### Project file & Debug log uploads

[full-parts.zip](https://github.com/user-attachments/files/19098535/full-parts.zip)

[debug_Wed_Mar_05_08_16_18_4026.zip](https://github.com/user-attachments/files/19098537/debug_Wed_Mar_05_08_16_18_4026.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

pm0u, 2025-03-05:
[image]

igiannakas, 2025-03-07:
It is because it is not a bridge, it's a wall. **A bridge is defined as the bridge infill that remains after the walls are printed.**

**Now how to get that printed with the correct speed.** In the external bridge speed tooltip explains that:
1. If you disable slow down for curled perimeters, the external bridge speed will be used to print overhang perimeters with overlap less than 13% -> so the bridge above falls under that category
2. If you enable slowdown for curled perimeters, the 75%-100% overhang speed is used.

There are good reasons to keep it like this - for example in organic models where the walls are partly overhanging, you want them printed as overhangs instead of bridges for best quality.

Hence why the options above are available and explained in the UI tooltips.

pm0u, 2025-03-08:
@igiannakas i appreciate you taking the time to respond but i do still believe this is a bug. Here is an almost identical feature also an "outer wall" on a model that is tilted 45deg similar to above

[image]

Perhaps bridge was the wrong term, but i believe this should be detected as an "overhanging wall"?

As it is in the original project/model, the slicer has generated a completely unprintable slicing profile and did not warn that the model would not be printable. I guess one or the other is the bug - the slicing generation or the lack of warning?

Realistically, the original issue should be able to be solved as the photo posted here - detecting the gap as an overhanging wall and not attempting to start and end the wall at the very edges of the existing wall?

This model in the latest photo was sliced with the same printing profile (don't slow down outer walls, slow down for overhangs, slow down layers to improve cooling).