OrcaSlicer/OrcaSlicer #8720: Printables integration broken on Flatpak

Coare1, 2025-03-06:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-beta2

### Operating System (OS)

Linux

### OS Version

Debian (Vanilla OS 2.0)

### Additional system information

CPU: Intel Core i5 12400f
GPU: AMD RX6600XT
RAM: 32gb DDR5

### Printer

Sovol SV08

### How to reproduce

1. Download and install 2.3.0-beta2 **flatpak** version
2. Navigate to Printables (on seemingly any web browser) and click on "Open in OrcaSlicer"

### Actual results

https://github.com/user-attachments/assets/a06688d1-17fc-47ae-b128-54bea9eaa2eb
Model will fail to import, giving the error "The file does not contain any geometry data".
[image]
Files will also be 0 bytes and will not be able to be opened at all.

### Expected results

Models should be able to be imported, and (presumably) show up as more than 0 bytes along with being able to be opened after the download.

### Project file & Debug log uploads

[failed_benchy.zip](https://github.com/user-attachments/files/19101370/failed_benchy.zip)

[debug_Wed_Mar_05_20_12_59_2.log.zip](https://github.com/user-attachments/files/19101372/debug_Wed_Mar_05_20_12_59_2.log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

I think this may be an issue with my operating system specifically, as I have had issues in the past with browsers and programs not communicating well, but if not it is most likely an issue with flatpak.

FredsMedia, 2025-03-07:
I have the same issue, running Arch Linux. It seems to also happen when importing any 3d model or project file, as well as the import feature. Downgrading to the 2.3.0 Beta flatpak gives the same result.

Coare1, 2025-03-07:
I can import 3d models, projects, and profiles fine, so that might be another issue. Does it give the same error?
Also I can confirm that it is broken with 2.3.0-beta flatpak.

FredsMedia, 2025-03-07:
Yes, the same popup appears.

MaUwJo, 2025-03-22:
I have the same issue - I guess it's something with network acccess as the files are zero bytes. - Not the import is failing - the download fails.
When I run this - I get the same error:
flatpak run --branch=master --arch=x86_64 --command=entrypoint --file-forwarding io.github.softfever.OrcaSlicer https://files.printables.com/media/prints/1211132/stls/9092110_d73d38b0-c0f7-4292-81b5-f41b679abffd_a4c17d85-c0d3-44c4-96fd-35af41c075d0/whistle_single_color.3mf

FredsMedia, 2025-03-22:
> I have the same issue - I guess it's something with network acccess as the files are zero bytes. - Not the import is failing - the download fails. When I run this - I get the same error: flatpak run [...]

The files were also 0 bytes for me, forgot to mention.

ShadowBr0ther, 2025-03-26:
same issue for me

FredsMedia, 2025-03-28:
Issue still persists on 2.30 Official Release

FredsMedia, 2025-06-27:
Issue is still a problem on latest version

NyremC, 2025-08-26:
I've had this issue also on my Linux / Bazzite OS, i had the issue suddenly after an upgrade using different versions of flatpak or Appimage. I also use an extra internal 4tb HDD for all my data, downloads, etc and that one was mounted in /var/mnt/. After trying to open the file from my boot drive it suddenly worked. Long story short i figured out that it was a **Permission** problem. In my case the error message was misleading, using Flatseal i gave permission to orcaslicer to access /var/mnt and it also worked on my data HDD. Maybe the devs could implement a permission check to show the proper error instead of "missing geometry data"

hope this helps anyone.

jounathaen, 2026-10-30:
The CLI issue seems to persist, but I cannot reproduce the original issue in 2.4.2, so probably this issue can be closed.

FredsMedia, 2026-10-30:
I can confirm this issue seems to be resolved.