OrcaSlicer/OrcaSlicer #8733: Error uploading to print host: HTTP 404

Beep6581, 2025-03-06:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

V2.3.0-beta2

### Operating System (OS)

Linux

### OS Version

Latest

### Printer

Creality Ender-3 V3 KE

### How to reproduce

Create a new project, and under the Printer panel click the Connection icon.

### Actual results

There seems to be no documentation explaining the "Host Type" options and which to select. The item selected by default was Octo/Klipper:

[image]

Note that the IP address is correct.

I found mention on one of Creality's pages that my printer uses firmware based on Klipper, so I thought the default setting of Octo/Klipper was correct. But when I click Test, I get a vague 404 error:

[image]

If I slice my project and click Print > Upload and Print, I get the same 404 popup.

However, if I go to Device, I can see my printer, control the fans and control XYZ. I can also import a g-code file, then click it and select "Start printing". So from a user's perspective, it looks like the "Connection" settings are correct but something is buggy.

I am stuck in a twilight zone where I Orca can actually see the printer and manually send print jobs, but "Print" doesn't work and leads to a mysterious 404.

### Expected results

To resolve this issue, document the Host Type options so that it is clear to the user which is correct. Further, there should be a clear way of testing that the settings are correct, and better feedback why "Print" fails.

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

rubienr, 2025-05-21:
With K2 i switched from `Creality Print` back to `Octo/Klipper` as `Host Type` since the upload rate is about 5.5 faster as with `Creality Print`.
See: https://github.com/orgs/CrealityOfficial/discussions/126#discussioncomment-13184323

Beep6581, 2025-11-24:
.