OrcaSlicer/OrcaSlicer #8735: CrealityPrint - Print does not work

Beep6581, 2025-03-06:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

v2.3.0-beta2

### Operating System (OS)

Linux

### OS Version

Manjaro

### Printer

Creality Ender-3 V3 KE

### How to reproduce

1. Create a new project, import an object.
2. Click the "Connection" icon in the Printer panel, which opens the "Physical Printer" window. Host Type is set to CrealityPrint. Even though I don't use [Creality Print](https://www.crealitycloud.com/downloads/software/creality-print), the "Test" is successful: [image] In fact, "CrealityPrint" is the only option which results in a successful test. Close "Physical Printer" window.
3. Slice plate.
4. Click Print > Upload and Print: [image]

### Actual results

The file uploads successfully and is visible in Device > File, but the top-left panel is named "Printing stopped" and nothing happens. [image]

If I click the file and select "Start printing" then it starts correctly.

So the bug is that when using "Host Type: CrealityPrint", the "Upload and Print" button uploads but does not print.

### Expected results

Uploads and prints.

### Project file & Debug log uploads

asd

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Related to #8733

Beep6581, 2025-04-08:
.

rubienr, 2025-05-21:
Off topic but try `Octo/Klipper` as `Host Type` instead of `Creality Print`. The upload rate is about 5.5 faster as with `Creality Print`. 
See: https://github.com/orgs/CrealityOfficial/discussions/126#discussioncomment-13184323

Beep6581, 2025-11-24:
.

Beep6581, 2026-10-18:
Still valid in 2.5.0-dev c1fb801f

brunoma-topocart, 2026-11-07:
I tracked down the cause on an Ender-3 V3 KE with OrcaSlicer 2.5.0 (host type CrealityPrint). The upload works, but the start command points to a path that doesn't exist, so the printer silently ignores it.

**What happens**

1. The upload to `/upload/<file>` succeeds. The file lands in `/usr/data/printer_data/gcodes/` (confirmed with `reqGcodeFile` over the websocket on port 9999).
2. `CrealityPrint::start_print()` builds the path from `is_k1_family(m_model)`:
   ```cpp
   const std::string data_root = is_k1_family(m_model) ? "/usr/data" : "/mnt/UDISK";
   ```
3. On the KE, `GET /info` returns:
   ```json
   { "mac": "...", "model": "F005" }
   ```
   `"F005"` is not in the `k1_models` set (`K1`, `K1 Max`, `K1 SE`, `K1C` and the `_CFS-C` variants), so Orca sends:
   ```json
   {"method":"set","params":{"opGcodeFile":"printprt:/mnt/UDISK/printer_data/gcodes/<file>"}}
   ```
4. The file actually lives in `/usr/data/printer_data/gcodes/<file>`. The printer never creates a job (`reqHistory` shows nothing for the uploaded file) and returns no error. Orca treats the K1-style connection close as success, so no error is shown either.

Starting the same file from the printer screen works normally.

**Suggested fix**

Add `"F005"` (Ender-3 V3 KE, which runs the same K1-style firmware: TEXT frames, `/usr/data` storage) to `k1_models` in `src/slic3r/Utils/CrealityPrint.cpp`. A more robust option would be to fall back to `/usr/data` when the printer's `reqGcodeFile` listing reports that path.

**Workaround** until then: use "Upload" only, then start the print from the printer screen or from Device → File.

b-amaral, 2026-11-07:
I tracked down the cause on an Ender-3 V3 KE with OrcaSlicer 2.5.0 (host type CrealityPrint). The upload works, but the start command points to a path that doesn't exist, so the printer silently ignores it.

**What happens**

1. The upload to `/upload/<file>` succeeds. The file lands in `/usr/data/printer_data/gcodes/` (confirmed with `reqGcodeFile` over the websocket on port 9999).
2. `CrealityPrint::start_print()` builds the path from `is_k1_family(m_model)`:
   ```cpp
   const std::string data_root = is_k1_family(m_model) ? "/usr/data" : "/mnt/UDISK";
   ```
3. On the KE, `GET /info` returns:
   ```json
   { "mac": "...", "model": "F005" }
   ```
   `"F005"` is not in the `k1_models` set (`K1`, `K1 Max`, `K1 SE`, `K1C` and the `_CFS-C` variants), so Orca sends:
   ```json
   {"method":"set","params":{"opGcodeFile":"printprt:/mnt/UDISK/printer_data/gcodes/<file>"}}
   ```
4. The file actually lives in `/usr/data/printer_data/gcodes/<file>`. The printer never creates a job (`reqHistory` shows nothing for the uploaded file) and returns no error. Orca treats the K1-style connection close as success, so no error is shown either.

Starting the same file from the printer screen works normally.

**Suggested fix**

Add `"F005"` (Ender-3 V3 KE, which runs the same K1-style firmware: TEXT frames, `/usr/data` storage) to `k1_models` in `src/slic3r/Utils/CrealityPrint.cpp`. A more robust option would be to fall back to `/usr/data` when the printer's `reqGcodeFile` listing reports that path.

**Workaround** until then: use "Upload" only, then start the print from the printer screen or from Device → File.

closed 2026-11-09 by raistlin7447