OrcaSlicer/OrcaSlicer #8736: Curb the bot

Beep6581, 2025-03-06:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

n/a

### Operating System (OS)

Linux

### OS Version

n/a

### Additional system information

- The github-actions bot spams issues with "stale", which adds nothing of value.
- Worse, it closes valid issues. Worst of all, we as the reporters are unable to re-open the closed valid issues and mark as still valid in the latest release. So the effort we put into clearly reporting the problem, into providing screenshots and steps to reproduce, is all for nothing.

That pisses your community off, and demotivates me/us from further feedback.

Please curb the bot.

### Printer

n/a

### How to reproduce

n/a

### Actual results

n/a

### Expected results

n/a

### Project file & Debug log uploads

n/a

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tlhintoq, 2025-03-07:
`Stale` is a good indication that you should make a remark on the thread saying it is still a problem.
Think of it as an early warning that in a week the thread will be closed.

If the problem is still a problem, just comment "still happening".

Beep6581, 2025-03-11:
What you're describing is project mismanagement, and it's ridiculous to have an issue irreversibly closed just because someone didn't keep bumping it every 7 days.

When one opens a new bug report for OrcaSlicer, the (annoyingly verbose) template asks:
> Is there an existing issue for this problem?
> Please search to see if an issue already exists for the bug you encountered.
> I have searched the existing issues

Why even bother, when the current setup encourages duplicates.

Ref: https://github.com/SoftFever/OrcaSlicer/commits/main/