OrcaSlicer/OrcaSlicer #8739: MCU 'mcu' shutdown: Move queue overflow

clutchkick512, 2025-03-06:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-beta2

### Operating System (OS)

Windows

### OS Version

windows 11

### Additional system information

cpu-13700k, 64gb ram,rtx 4080

### Printer

k1

### How to reproduce

load up any round object, slice with archane walls and print. 

### Actual results

I have been able to repeate the failure 3 times in a row with cylinders about 65mm wide and 100mm tall. near the 85mm mark the printer has the MCU 'mcu' shutdown: Move queue overflow error and stops klipper.

### Expected results

the file should print without issue.

### Project file & Debug log uploads

I am unable to provide this as it is a paid file. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I am a support mod for the k1 software SimpleAF and have had at least 1 other user on the beta report the same issue. In my case turning wall generation to classic and disabling arc fitting resolved the issue. I beleive the exported gcode files may be too large for the k1 memory but on the latest stable release it is not an issue at all.  I have rolled back to the stable as has the other discord member of SimpleAF and fixed the issue with that.

igiannakas, 2025-03-07:
Try adding the below in your start g-code:
M413 S0 

This **disables power loss recovery** which places significant compute load on the K1 CPU. 

Also **you MUST disable arc fitting for the K1 and all klipper machines** as they are not using these commands natively - instead they convert them internally back into line segments. This results in both a worse print quality and also increased CPU load.

Also make sure you have **verbose g-code output set to disabled in Orca**.

Finally do not use the extrusion rate smoothing feature as this generates smaller line segments to enable gradual slow downs and speed ups. While not being able to use this will affect overhang quality, if the printer cannot cope with the gcode load, then it's best disabled. However, optimisations in 2.3 have been also done for this feature: https://github.com/SoftFever/OrcaSlicer/pull/7398

A more drastic way is to also reduce the precision -> resolution setting but again this may affect print quality if you go too high.

The issue stems from the fact that the CPU used by creality is too slow to keep up with the speed the printer is commanding it to go.

There have been several fixes already in 2.2 and 2.3 to reduce gcode size and optimise it, especially for overhangs (eg: https://github.com/SoftFever/OrcaSlicer/pull/6714), but there is only so much that can be done.

clutchkick512, 2025-03-07:
Interesting, so just for notes SimpleAF doesn’t have the power loss recovery already because of the CPU load.

Disabling arc fitting- does this also change spiral z hop back to normal or slope? That’s about the only real arc fitting feature I use.

igiannakas, 2025-03-07:
There are two places where arcs are used.
1. Arc fitting in the quality tab. This creates arcs for extrusion moves (ie printing). This should be disabled.
2. spiral Z hop: this creates arcs for non printing moves (travel). These can be enabled on a klipper machine (and should be to be honest as the quality benefit is significant).

My recommendation is to leave spiral z on and arc fitting off. However if your printer cannot cope with the spiral z speed you can increase the arc segment in klipper itself (not sure if the creality machine allows you to have access to the printer cfg) or simply use slope z hop.

Hope this helps!

bpengu1n, 2025-09-29:
I'm having this come up as well on a print with many circles. If I reduce the number of those objects (from 18 to 9 oval pieces with approx 6 circles each), I don't get the failure. I thought it might be an issue with my printer (Sovol SV08), but after following their procedure to fix I still have this issue.

evelant, 2025-12-05:
@bpengu1n I'm having the same issue on an SV08. Did you ever get it figured out?

bpengu1n, 2025-12-06:
There are a number of things that could cause this, but the fix in my case was to modify one of the Klipper core files, mcu.py, and change the MCU comm timeout from 0.025 to 0.05. I haven’t seen the issue a single time since making this change (I also since went to mainline Klipper and installed Kalico which supports changing this timeout in the user config).

I don’t remember the guide I followed, but this page gives an easy one-line command to make the change when you SSH into your printer: https://mellow.klipper.cn/en/docs/DebugDoc/faq/hometimeout/

(Edit: the post I found with the steps originally was https://forum.sovol3d.com/t/console-error-communication-timeout-during-homing-probe/8276. I highly recommend backing up this file before changing it)

evelant, 2025-12-06:
For anybody else who gets stuck with this error I found the root cause and figured out how to fix it. The problem is Sovol’s PLR (power loss recovery) customization to Klipper. This is why moving to mainline Klipper fixes it, because that removes Sovol’s PLR customization. It’s pretty easy to fix on the Sovol firmware now that we know what’s causing it.

The problem is in ~/klipper/klippy/extras/gcode_move.py [Lines 123-133](https://github.com/Sovol3d/SV08/blob/a60644875f8c756d20b3828c9416518b414b5491/home/sovol/klipper/klippy/extras/gcode_move.py#L123-L133). These lines write information to a file on the sd card synchronously on every single move command. That’s too slow, especially for python on a relatively constrained host. It causes moves to be processed too slowly when there’s a lot of short moves, the host fails to communicate with the mcu fast enough resulting in the move queue overflow error.

To fix it, ssh into the printer and simply comment out these lines in ~/klipper/klippy/extras/gcode_move.py

        if 'G' in params and 'Z' in params and 'X' in params and 'Y' in params and 'F' in params:
            if self.v_sd.cmd_from_sd:
                #commandline = gcmd.get_commandline()
                content = {
                    'commandline': gcmd.get_commandline(),
                    'Z': params['Z'],
                    'extrude_type': 'M82' if self.absolute_extrude else 'M83',
                    'e_extrude_abs': 0 #self.Coord(*self.move_position)[3]
                }
                with open("/home/sovol/sovol_plr_height", 'w') as height:
                    json.dump(content, height)

PLR will obviously no longer work, but it never worked well to begin with and failed every single time when the move queue overflow happened since it was the writing of plr info causing the failure. Hopefully Sovol fixes this in an official firmware update. It was hard to track down but in hindsight it’s a pretty obvious huge flaw in the PLR design – there’s no way that writing to a file synchronously with every move command is going to work reliably.

Now I can print as fast as I want with spiral z-hop and organic supports, no more problems!

clutchkick512, 2025-12-06:
> For anybody else who gets stuck with this error I found the root cause and figured out how to fix it. The problem is Sovol’s PLR (power loss recovery) customization to Klipper. This is why moving to [...]

Yeah not the actual problem though. For creality machines on SimpleAF we found it was a mishandled m106 macro. SimpleAF doesnt hav ethe PLR code anymore and we were still seeing the move queue overflow errors.

evelant, 2025-12-07:
Sorry, I edited my post to clarify that at least on my sovol sv08 this was the source of the problem and there may be similar "features" on other printers triggering the same failure. My hope is that it at least gives people a clue to help diagnose the issue.

drewslarsen-hash, 2026-01-15:
i have an ender 3v3 plus and ive connected to the printer with ssh but it seems mcu.py is read only and im unable to modify it in any way. anyone got any ideas?

igiannakas, 2026-05-02:
Indeed the micro step count is ridiculously high for the hardware they are using (mcu). Also there is zero precision benefit going that high. 

On my high speed Voron I’m using 32 microsteps and interpolation.

SnifMyBack, 2026-05-02:
> Also as per klipper documentation hold current should equal run current.

So before seing your answer, I've tried the 32 micro steps with interpolation (even though some people say that it affect accuracy) and it worked like a charm. I feel so stupid to have overlooked this while digging deeper in python scripts related to the CFS integration.

Many thanks, you saved me a lot of debugging time.