OrcaSlicer/OrcaSlicer #8747: No custom filament to load to device

Melanie-Hesse, 2025-03-07:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 beta 2

### Operating System (OS)

macOS

### OS Version

macOS 15.3.1

### Printer

Bambu Lab A1 - with AMS lite

### How to reproduce

1. go to devices tab
2. click on pen to edit filament
3. select loaded filament

### Actual results

There is no custom filament to select, only all standard filament.

### Expected results

All custom filament are selectable.

### Project file & Debug log uploads

[Archiv.zip](https://github.com/user-attachments/files/19125231/Archiv.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Melanie-Hesse, 2025-03-07:
Maybe I do need to specify more precisely. 
I already added about 25 custom filaments the way mentioned by @Schildkroet. That was in January and worked quite fine until I did the update to 2.3.0. Now there are no more custom filaments in the devices-tab of Orca to choose for the AMS lite.

Schildkroet, 2025-03-07:
Just checked it. Can confirm that is is not possible to select any custom filament in AMS.

Melanie-Hesse, 2025-03-07:
Just found ONE custom filament, but this one I created after installing the update.

sucaba, 2025-04-01:
In my case there is only one of many custom filaments which is not shown in the AMS dropdown. 
And when I opened info about it in the list of custom filments I found that it lacks some information:
[image]

Though, in the full filament details dialog vendor name is available:

[image]