OrcaSlicer/OrcaSlicer #8754: Weird little lines appearing

TheUnlimited64, 2025-03-07:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-Beta2

### Operating System (OS)

Windows

### OS Version

11

### Printer

Bambulab P1P

### How to reproduce

https://www.printables.com/model/229538-print-in-place-bag-clip-2-parametric

Slice this model with 0.6mm nozzle. A weird line appears

[image]

[image]

### Actual results

A weird little line gets sliced

### Expected results

Line does not appear

### Project file & Debug log uploads

[Bug project.zip](https://github.com/user-attachments/files/19131935/Bug.project.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

discip, 2025-03-08:
Just set **`Brim type`** to **`No-brim`**. 😊

[image]

TheUnlimited64, 2025-03-09:
Ah lol, weird that thats the brim. Is that expected behavior?

closed 2025-03-09 by TheUnlimited64