OrcaSlicer/OrcaSlicer #8759: Keeps forcing itself to foreground

officiallymarky, 2025-03-08:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

Arch

### Additional system information

KDE Wayland Nvidia

### Printer

Bambu X1C

### How to reproduce

1. Print
2. Minimize Orca Slicer

### Actual results

Orca Slicer keeps forcing itself to the foreground every update from the printer, typical updates filament change, stage changes (finished bed leveling), etc.  Printing one color, it only happens during the start of the print, then will behave until the end of the print.

### Expected results

If i minimize Orca Slicer, it should stay minimized, except maybe on error.

### Project file & Debug log uploads

Nothing specific in logs that relate

### Checklist of files to include

- [x] Log file
- [ ] Project file

### Anything else?

I frequently will print something, and minimize orca while I do other things, sometimes playing games.  This constant forcing to the foreground is very disruptive and forces me to close orca when I don't really want to close it (especially when a printer is multiple plates and I want to save what I'm doing).

tlhintoq, 2025-03-14:
Just a thought... 
Don't go to the `Device` tab after sending the job.
See if it keeps jumping to foreground if you go back to `Prepare` after sending the job.

gray-space, 2025-03-30:
> Just a thought... Don't go to the `Device` tab after sending the job. See if it keeps jumping to foreground if you go back to `Prepare` after sending the job.

It only pops to the foreground on the device tab. But that's not a sufficient workaround. That means to take a quick peek at the print in progress, you need to bring the window to the foreground, click the Device tab, wait for the video to start streaming again, an finally look. Then you have to remember to click the Prepare tab before  minimizing again. That's a lot more work than just bringing the window up to look at it and then minimize again. It also makes it impossible to use OS features like hovering over the icon in the taskbar/dock to be able to see the thumbnail preview of the window. That's often good enough to make sure the print hasn't gone horribly wrong.

Waltibaba, 2025-06-04:
Same problem here.
This issue was fixed in Bambu Studio in February: https://github.com/bambulab/BambuStudio/issues/3046

closed 2025-09-10 by github-actions[bot]