OrcaSlicer/OrcaSlicer #8774: Hide modifiers for tools like seaming and support painting

tlhintoq, 2025-03-09:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Sometimes you realize you need to paint seams after you've put in modifiers... But you can't paint through the modifier in some cases.

It would be really helpful if tools such as support painting, and seam painting hid modifier objects.

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Please hide modifiers when using painting tools.

Noisyfox, 2025-03-09:
> But you can't paint through the modifier in some cases.

Why not?

tlhintoq, 2025-03-10:
I swear it wasn't working yesterday: It kept switching selection from the object to the modifier.  Today its fine and making a liar out of me.
Hey - okay.  It works.  My bad.

closed 2025-03-10 by tlhintoq

Noisyfox, 2025-03-10:
> It kept switching selection from the object to the modifier.

That's something I noticed too and has raised PR #8776

tlhintoq, 2025-03-10:
> > It kept switching selection from the object to the modifier.
> 
> That's something I noticed too and has raised PR [#8776](https://github.com/SoftFever/OrcaSlicer/pull/8776)

Oh.  Ok.  Glad someone else saw it because I thought I must have just been stupid-tired or something and it was operator error