OrcaSlicer/OrcaSlicer #8787: Pressure advance batch mode testing inputs not appearing

justsayrow1, 2025-03-10:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-beta2 build 93cb042

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Sovol SV06 ACE

### How to reproduce

1. Open a new project
2. Select "Calibration" on the top bar
3. Select "Pressure Advance"
4. Select "DDE" and "PA Pattern"

### Actual results

Inputs for Accelerations and Speeds for the purposes of batch mode testing do not appear in the PA Calibration configuration window

[image]

### Expected results

The PA Calibration configuration window should have input boxes for Accelerations and Speeds as show in [adaptive pressure advance](https://github.com/SoftFever/OrcaSlicer/wiki/adaptive-pressure-advance) and [Calibration](https://github.com/SoftFever/OrcaSlicer/wiki/Calibration)

[image]

### Project file & Debug log uploads

[PA Issue Project.zip](https://github.com/user-attachments/files/19157528/PA.Issue.Project.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

brocc28, 2025-03-10:
Same issue
OS: MacOS 15.3 (24D60)
Architecture: ARM - Apple Silicon M1 Pro
Orca Version: V2.2.0 Official Release

[image]

justsayrow1, 2025-03-10:
Updating to nightly build 2.3.0-beta2 c17b52a appears to have fixed the issue 

[image]

closed 2025-03-10 by justsayrow1