OrcaSlicer/OrcaSlicer #8794: Interface shells should not apply to bottom surfaces

IanVS, 2025-03-10:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-beta2

### Operating System (OS)

macOS

### OS Version

15.3.1

### Printer

Bambu P1P

### How to reproduce

Create a model with a multi-material transition on a bottom surface.  Enable interface shells.  When sliced, you'll see that the material on top is segmented, which will cause problems because of gravity.

### Actual results

Here are some screenshots of what happens:

[image]

[image]

### Expected results

This is what I get without interface shells.  I would like something like this, but ideally without the artifacts in the centers.  

[image]

### Project file & Debug log uploads

[InterfaceShellsIssueExample.3mf.zip](https://github.com/user-attachments/files/19164147/InterfaceShellsIssueExample.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

I've uploaded an example project that's a bit simpler.  This is what it looks like with interface shells enabled:

[image]

IanVS, 2025-03-10:
As a workaround, it would be great if interface shells could be specified in a layer modifier.

Twenty5Schmeckels, 2025-05-02:
> As a workaround/fix, it would be great if interface shells could be specified in a layer modifier.

This would actually solve a few other issues with interface shells I've encountered, would love to see that as an option!